motioneye-project/motioneye #1378: Two usb cams with motioneye 0.4 do not work

Honei12, 2019-07-24:
When i connect two cameras to the nano pi motioneye always display one at a time.

I already have tried to change the video_device to /dev/video10 and /dev/video8.

The command "lsusb" shows this:

Bus 004 Device 004: ID 15aa:1555 Gearway Electronics (Dong Guan) Co., Ltd.
Bus 004 Device 003: ID 15aa:1555 Gearway Electronics (Dong Guan) Co., Ltd.
Bus 004 Device 002: ID 2109:0817 VIA Labs, Inc.
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2109:2817 VIA Labs, Inc.
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The command "v4l2-ctl --list-device" shows this:

rkisp1-statistics (platform: rkisp1):
        /dev/video2
        /dev/video3
        /dev/video6
        /dev/video7

rkisp1_mainpath (platform:ff910000.rkisp1):
        /dev/video0
        /dev/video1

rkisp1_mainpath (platform:ff920000.rkisp1):
        /dev/video4
        /dev/video5

3.0 USB Camera (usb-xhci-hcd.11.auto-1.2):
        /dev/video10
        /dev/video11

3.0 USB Camera (usb-xhci-hcd.11.auto-1.4):
        /dev/video8
        /dev/video9

Log file:

[-1630226544:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /home/pi/motioneye/motion.conf
[-1630226544:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[-1630226544:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "vid_control_params"
[-1630226544:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
[-1630226544:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "vid_control_params"
[-1630226544:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
[-1630226544:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
[-1630226544:motion] [NTC] [ALL] motion_startup: Logging to syslog
[-1630226544:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[-1630226544:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[2:ml2:Camera2] [ERR] [VID] v4l2_select_input: Unable to query input 0. VIDIOC_ENUMINPUT, if you use a WEBCAM change input value in conf by -1: Inappropriate ioctl for device
[1:ml1:Camera1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2:ml2:Camera2] [ERR] [ALL] motion_init: Error capturing first image
[2:ml2:Camera2] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[2:ml2:Camera2] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[2:ml2:Camera2] [ERR] [VID] v4l2_select_input: Unable to query input 0. VIDIOC_ENUMINPUT, if you use a WEBCAM change input value in conf by -1: Inappropriate ioctl for device
[2:ml2:Camera2] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[2:ml2:Camera2] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

The problem here is, when i try to change input to -1 it doesn't recognize it.

efogli, 2022-12-15:
> When i connect two cameras to the nano pi motioneye always display one at a time.
>
> I already have tried to change the video_device to /dev/video10 and /dev/video8.
>
> The command "lsusb" shows this: [...]

Hi Hannes,
I facing the same problem using motion with two identical HP webcam HD 2300.
Did you solve it in some way?

Thanks
Efogli

Phoenix1747, 2023-01-11:
Same problem with 2 VGA capture cards. Any clues?

Can you do a `lsusb` and see what the usb ids are? I suspect a problem with the usb product ids not being unique.