motioneye-project/motioneye #1379: Not able to select camera while adding a network camera

sYnHybrid, 2019-07-25:
Try to add a MJPEG camera to my motioneye Docker master-amd64.
I have access to the camera and the right credentials, but after entering the credentials the dropdown menu to select a camera just vanishes without an error.
[image]
Started a new container in Host-Mode and the problem still exists.
I installed MotionEyeOS on a RaspberryPi3 same Problem and I can't add the Raspberry Pi stream to the Dockercontainer with the same nonexisting dropdown menu.

Any ideas or people with the same problem?

sYnHybrid, 2019-07-25:
on an old install with raspbian 9.6, motioneye 0.39.3 and motion 4.1.1 it works but the docker still does not display the camera option.
[image]

subnet, 2019-08-21:
Hello,
same thing for me. As you can see in attached image, I can't select camera in dropdown menu, and no down arrow is present.
Installed on Raspbian Buster with wiki's instructions. MotionEye version 0.40 Motion 4.2.2 OS Raspbian 10

[image]

ccrisan, 2019-08-23:
@sYnHybrid please attach your motioneye log file.
@subnet you haven't entered the camera URL.

subnet, 2019-08-24:
thanks for your answer, now it seems to work. I was able to add two sv3c ip IP camera and an old Rasberry with a raspicam running v4l2rtspserver
Thanks a lot for your work :-)

gleitoncampos, 2024-10-09:
Same problem  here... No options in drop down selection for the type for the camera. Using docker installation and no useful logs in output

[image]
[image]
[image]

dihan, 2024-10-19:
same issue as @gleitoncampos 

> motionEye Version | 0.42
> -- | --
> Motion Version | 4.2.2+gitUNKNOWN
> OS Version | motionEyeOS 20200203

JamBalaya56562, 2026-11-07:
Closing as answered: as dihan found above, the camera list stays empty when the user name and password are part of the RTSP URL. Enter the URL without them and put them in the user name and password fields.

closed 2026-11-07 by JamBalaya56562