motioneye-project/motioneye #1380: Frame Change Threshold setting 0%

2boot, 2019-07-25:
On installation of MotioneyeOS the default setting of "Frame Change Threshold" appears to be ZERO%, after moving the setting to another value the default "zero" remained in my mind, so I returned it to zero along with other settings updated at the same time. From then on I did not get any motion recorded.

I have searched FAQ and can see other complaints apparently without solution.
So I would like to put on record the slider although appearing at zero is actually around 0.9%, just click the slider and move the tiniest amount and the value shows on screen.
So set it to 0% and it will not detect any motion which is far from my assumption that it would be the utmost sensitivity.
#jasaw did ask for a person to test this setting but originator response to that issue was incomplete and issue was subsequently closed.

This issue may be closed immediately - thanks

prb28, 2020-01-12:
I had the same behavior, with motioneyeos 20190911. This issue helped me to figure out to get the motion detection back, thanks.

2boot, 2020-01-12:
Thank you for posting your response, It's nice to know that I have helped.

2boot, 2020-01-12:
Maybe more persons would read the details if I marked it CLOSED myself

closed 2020-01-12 by 2boot