motioneye-project/motioneye #1382: My method of Uploading to OneDrive using RCLone with MotionEye trigger

JBlyth, 2019-07-26:
Is this the right place for this information????

I might be teaching everyone how to suck eggs, but I have been able to get MotionEye to upload successfully to OneDrive using RClone on my Raspberry Pi3+. My script uploads the last two days of motion detections and only uploads files that are not present on OneDrive or have their file size changed for some reason, so it is quite quick. It looks back for 20 days in case of lack of alarms. This is done for speed. It also exits if the script is already running, but this may not be required....

In MotionEye, use a "Run an End Command" pointing to your script where the last word is the name of your camera:- /etc/motioneye/MotionEyeToOneDrive.sh Garage

On your MotionEye Device, create a folder /home/pi/RClone for log files

In the /etc/motioneye/ folder, create a bash script along the lines of the following. Edit the first lines to suit as commented  :-

#!/bin/bash
requiredUser="pi" #Edit to suit username of the RCLone config
remoteFolderName="OneDrive:SecurityCameraPi" #Edit to suit your OneDrive remote folder
remoteFolder=$1
if [[ $remoteFolder == "Garage" ]]; then #Edit to suit your camera name in MotionEye
 localFolder="Camera1" #Edit to suit you local camera folder
fi
if [[ $remoteFolder == "House" ]]; then  #Edit to suit your camera name in MotionEye
 localFolder="Camera2" #Edit to suit you local camera folder
fi
#### Add more cameras here as required
if [[ $remoteFolder == "" ]]; then
 echo "ERROR, ROOT Folder" $rootFolder "NOT FOUND" > /home/pi/RClone/Error.txt
 exit
else
 rm -f /home/pi/RClone/Error.txt
fi
DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"  # get cur dir of this script
progName=$(basename -- "$0")
cd $DIR
if pidof -o %PPID -x "$progName"; then
 echo "WARNING  - $progName Already Running. Only One Allowed." >> /home/pi/RClone/Upload.txt
 echo "WARNING  - $progName Already Running. Only One Allowed."
else
 num=0
 while [ $num -lt 20 ]
 do
  localSubDirName=$(date -d @$(( $(date +"%s") -($num*(24*3600)))) +"%Y-%m-%d")
  localCamFolder="/var/lib/motioneye/"$localFolder"/"$localSubDirName
  remoteCamFolderName=$remoteFolderName"/"$remoteFolder"/"$localSubDirName
  #echo $localCamFolder "to" $remoteCamFolderName
  if [ ! -d "$localCamFolder" ] ; then   # Check if Local sync Folder Exists
   echo $localCamFolder "Folder NOT Found " $num
   ((num=num+1))
  else
   nowDay=$(date +"%d")
   nowSuffix="th"
   if [ $nowDay -eq 1 ]; then
    nowSuffix=$nowDay"st"
   fi
   if [ $nowDay -eq 21 ]; then
    nowSuffix=$nowDay"st"
   fi
   if [ $nowDay -eq 21 ]; then
    nowSuffix=$nowDay"st"
   fi
   nowDate=$(date +"%A,%d$nowSuffix %B %Y")
   logDate=$(date +"%d-%m-%Y")
   nowTime1=$(date +"%T")
   #echo "Upload Started on" $nowDate "at" $nowTime1
   #echo $num "- Sending" $localCamFolder "to" $remoteCamFolderName
   sudo -u $requiredUser rclone copy --log-file /home/pi/RClone/RCloneLog.txt --size-only --exclude *.thumb $localCamFolder $remoteCamFolderName
   nowTime2=$(date +"%T")
   #echo "Upload Started on" $nowDate "at" $nowTime1 "- FINISHED AT" $nowTime2
   echo  $logDate "Started" $nowTime1 "Finished" $nowTime2" >> /home/pi/RClone/Uploads.txt
   echo $localCamFolder "to" $remoteCamFolderName >> /home/pi/RClone/Uploads.txt
   ((num=num+1))
   if [ $num -eq 2 ]; then
    break
   fi
  fi
 done
fi
echo "Finished after" $num "folders"

FabriceIAD, 2021-02-14:
Hello, I don't see where is the onedrive account setup please?

starbasessd, 2021-02-14:
There isn't one like Dropbox or GDrive.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

fgrosswig, 2021-08-02:
> Hello, I don't see where is the onedrive account setup please?

https://rclone.org/onedrive/ and https://rclone.org/downloads/ for the tools on any plattform to continue headless if needed.

starbasessd, 2021-08-02:
Onedrive is not an option, only GDrive and Drop box.

fgrosswig, 2021-08-02:
> Onedrive is not an option, only GDrive and Drop box.
> […](#)

wasnt asked about an "option" but rclone has the ability for onedrive 😄 so if a user preferes onedrive he could manage it by the shell, no ❓

starbasessd, 2021-08-02:
No clue. You might want to tag @JBlyth since that is his setup script.

JamBalaya56562, 2026-11-10:
Closing, as this is a write-up rather than an issue: uploading to OneDrive with rclone from Run An End Command. Thanks for sharing it; it stays here for others to find.

closed 2026-11-10 by JamBalaya56562