motioneye-project/motioneye #1383: Installing on Raspbian, package not found

darkplatinum, 2019-07-26:
Hello, when I tried to install libmariadbclient18 on the latest version of raspbian (buster) I get that the package is not available and libmariadb3 will be replaced with it. Is this ok?

zagrim, 2019-07-27:
The instructions for Raspbian installation are for Stretch and not yet updated/appended for Buster, where things seem to go a little differently.
There are a couple of issues on the subject already: 
- #1356 (links to a script which might be useful, although it still downloads `motion` from github and doesn't use the package from Buster repos)
- #1348 

Basically, you might have less conflicts if you used the Raspbian version of `motion`, at least. I'm not actually sure why MariaDB and Postgres are listed as dependencies, but go with what Buster provides for those and you'll probably be fine.

It would be great if you, or someone else, could draw together updated installation instructions on Raspbian Buster, so a maintainer could update the Wiki :smile:

darkplatinum, 2019-07-27:
> It would be great if you, or someone else, could draw together updated installation instructions on Raspbian Buster, so a maintainer could update the Wiki 😄

I might actually do that. I have a Raspberry Pi tutorial website as well as a small youtube channel so you never know...

darkplatinum, 2019-07-27:
Also one more thing thats off topic, python-dev installs python 2.7. Will this be updated to a newer version of python as python EOL is next year?

darkplatinum, 2019-07-27:
hmm..... I followed those links and installed motioneye. Unfortunately when I try to use the MMAL camera on the raspberry pi it just shows different colors  like green, purple etc.

darkplatinum, 2019-07-27:
[image]

zagrim, 2019-07-28:
Upgrade to support Python 3 in motionEye has been planned and there's been at least on PR already for that, but I don't know when will it actually be included in a release.

Unfortunately I cannot give any definitive advice on the camera issue. You might try running `motion` alone and either configuring it (edit `/etc/motioneye/camera-*.conf` or whatever you have) with `emulate_motion on` and check if the produced video files are ok, or with `stream_port` (and perhaps `stream_localhost`) so that you can view the camera stream in browser. (see [motion config guide](https://motion-project.github.io/motion_config.html) for details on the config parameters)

iBlocker-eu, 2019-07-28:
Hi, you can install on Buster it using the Raspbian instructions, replace libmariadbclient18 with libmariadb3, and it worked for me. The problem is that I have declared a camera but maximize button is not working, Preferences menu is sometime not accessible, the same as Apply button, which is not always visible after modifying settings.

darkplatinum, 2019-07-28:
> Hi, you can install on Buster it using the Raspbian instructions, replace libmariadbclient18 with libmariadb3, and it worked for me. The problem is that I have declared a camera but maximize button [...]

I've tested this out yesterday and had so many problems with it. Masking not working and problems stated above. So I decided to use MotioneyeOS which works straight out of the box with extra functionality over motioneye.

KarlBaumann, 2019-07-29:
Please resolve this issue with high priority

cbytestech, 2020-01-31:
> I've tested this out yesterday and had so many problems with it. Masking not working and problems stated above. So I decided to use MotioneyeOS which works straight out of the box with extra [...]

I just had good luck with this
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
if it runs into an unmet dependancy error, run the apt -fix command it gives you!! dont ignore it.
Then just go step by step. I believe in you! You got this bruh!

cbytestech, 2020-01-31:
Also..to resolve the issue they dont mention in the tutorial...
https://www.raspberrypi.org/forums/viewtopic.php?t=25611

make sure that only viewing as localhost is off in both 
`sudo nano /etc/motion/motion.conf` and 
`sudo nano /etc/motioneye/motioneye.conf`
change
webcam_localhost off
control_localhost off

it doesnt hurt to look into the other settings in there as well.

closed 2020-09-29 by RaspberryTech01