motioneye-project/motioneye #1386: Virtual Maschine - No image/stream from/to motioneye

JBSullivan, 2019-07-28:
Hello, everybody,
First, the facts. I already use MotioEye on a Raspberry with two cameras. This one works almost perfectly. Now I wanted to extend it with two more cameras. So the Raspberry reaches its performance limit.

I then set up a virtual machine with VirtualBox on my Smarthome server.

This virtual machine runs Debian Linux and MotionEye. The network connection from the VM to the host PC is set up as a bridge.

motionEye Version 0.40
Motion version 3.2.12+git20140228
OS version Linux 3.16.0-4-586

On the local PC, where also the VM is running, I can get to MotionEye via browser as usual.

However, if I enter the same camera parameters (e.g. network camera / rtsp://10.xxx.xxx.xxx.xxx:10554/tcp/av0_0), as they work successfully on the Raspberry, I only get an empty image/stream.

The interesting thing is - if I call the IP address of the MotionEye VM session on another PC in the same network, I get MotionEye, but only the "preferences" tab is displayed.

I already set the ssh root rights for remote connections to YES. But that doesn't make any difference. I just can't get a connection to the camera streams which should run on the VM in the MotionEye session.

Does anyone have a smart idea?

Translated with www.DeepL.com/Translator

JBSullivan, 2019-07-28:
Now i generate a LOG file - but can not interprete this data.

---------------------------------------------------------------------------------------------------------

`-- Logs begin at So 2019-07-28 17:16:06 CEST, end at So 2019-07-28 17:19:00 CEST. --
Jul 28 17:16:12 debian meyectl[345]: INFO: hello! this is motionEye server 0.40
Jul 28 17:16:12 debian motion[454]: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Jul 28 17:16:12 debian motion[454]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
Jul 28 17:16:12 debian motion[454]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "camera"
Jul 28 17:16:12 debian motion[454]: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Jul 28 17:16:12 debian motion[454]: [0] [NTC] [ALL] motion_startup: Logging to syslog
Jul 28 17:16:12 debian motion[454]: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jul 28 17:16:12 debian motion[454]: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jul 28 17:16:12 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:16:12 debian motion[454]: [1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
Jul 28 17:16:14 debian meyectl[345]: ERROR: could not find motion camera id for camera with id 3
Jul 28 17:16:14 debian meyectl[345]: ERROR: could not find motion camera id for camera with id 5
Jul 28 17:16:14 debian meyectl[345]: INFO: cleanup started
Jul 28 17:16:14 debian meyectl[345]: INFO: wsswitch started
Jul 28 17:16:14 debian meyectl[345]: INFO: tasks started
Jul 28 17:16:14 debian meyectl[345]: INFO: mjpg client garbage collector started
Jul 28 17:16:14 debian meyectl[345]: INFO: server started
Jul 28 17:16:14 debian meyectl[345]: ERROR: failed to disable motion detection for camera with id 1: Not Valid
Jul 28 17:16:20 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:16:20 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:16:30 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:16:30 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:16:40 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:16:40 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:16:50 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:16:50 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:17:00 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:17:00 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:17:10 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:17:10 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:17:20 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:17:20 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:17:30 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:17:30 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:17:40 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:17:40 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:17:50 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:17:50 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:18:00 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:18:00 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:18:10 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:18:10 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:18:20 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:18:20 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:18:30 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:18:30 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:18:40 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:18:40 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:18:50 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:18:50 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Jul 28 17:19:00 debian motion[454]: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jul 28 17:19:00 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:`

zagrim, 2019-07-28:
> Jul 28 17:16:12 debian motion[454]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
> Jul 28 17:16:12 debian motion[454]: [0] [ALR] [ALL] conf_cmdparse: Unknown config option "camera"

It looks like your config is for `motion` version 4.x but you are running with version 3.2. Just upgrade the "motion" package to newer version. You are not saying what Debian you are running on, but Stretch has 4.0 which should be better match. Alternatively, you could download .deb file from https://github.com/Motion-Project/motion/releases/ for whatever Debian release you have.

JBSullivan, 2019-07-28:
I think this is the main reason for my image problem.

> Jul 28 17:16:12 debian motion[454]: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:

Is it right that this code only works on a Raspberry withe a Raspberry cam? I`m using a VM without any local cams.

JBSullivan, 2019-07-28:
I'm using Debian 8.

I am not so familiar with Linux, but upgrading to motion 4.2.2 will not work. I think it's my bug - but where is the bug?

wget https://github.com/Motion-wgetProject/motion/releases/jessie_motion_4.2.2-1_i386.deb

zagrim, 2019-07-29:
I don't know of your cam, but if it is a network camera, I can see no direct reason why it wouldn't work, only if Motion could find it's configuration. Motion 4.x uses config option "camera" to point to individual camera config files from the main `motion.conf`, which isn't recognised in your case, so it doesn't even know of your camera, I believe.
Your download link is incorrect, copy-pasted from the github Releases page for Debian 8 i386:
https://github.com/Motion-Project/motion/releases/download/release-4.2.2/jessie_motion_4.2.2-1_i386.deb

JamBalaya56562, 2026-11-08:
Closing, as zagrim found the cause above: Motion 3.2.12 on Debian 8 does not know the `camera` option that motionEye 0.40 writes into `motion.conf`, so Motion never saw the camera. motionEye needs Motion 4; today `motioneye_init` installs a current Motion package, see the [README](https://github.com/motioneye-project/motioneye#installation).

closed 2026-11-08 by JamBalaya56562