motioneye-project/motioneye #1387: Action Button via URL from Server to the GPIO pin on network camera.

HeAnVu75, 2019-07-28:
Hi, Calin Crisan.

I want to thank you for making MotionEyeOS. It is a brilliant CCTV operating system and APP.
I have read through Faq and have donated to your project.
In my CCTV system I have one Raspberry PI 3B + as server / HUB and four Raspberry PI Zero each with their own NoIR Camera version 2 as network camera. What I need to learn is how to enable GPIO pin on one of the PI Zero through the server PI 3B + by using the Action Button? I want the Server to handle motion detection and store images on external disk. And also control action buttons eg. light_on_4 and light_off_4 on the fourth Raspberry Pi Zero network camera. How can I use bash script commands to control the GPIO pin on Pi zero through the server? Think it has to do with URL, IP addresses, user and password.

Sincerely, HeAnVu75

hexordecimal, 2019-11-26:
python scripts, look up gpizero and pigpio. then place those scripts in /etc/motioneye/ and sudo chmod +x down_1 for each.

JamBalaya56562, 2026-11-08:
Closing as answered above. When a camera is added from another motionEye server, its action buttons are passed on to that server, so the scripts belong on the Pi Zero. Put them in its configuration directory, named after the camera id there, e.g. `light_on_1`, and make them executable. See [Action Buttons](https://github.com/motioneye-project/motioneye/wiki/Action-Buttons).

closed 2026-11-08 by JamBalaya56562