motioneye-project/motioneye #1388: RPi Buster - ModuleNotFoundError

testbughub, 2019-07-29:
Hi!

I just reinstalled my RPi server with Buster, which I know isn't officially supported yet,
but I'm getting an error I've not seen in the Issues tab so I thought I'd just report it.
If it's unnecessary I or someone else can just close this.

Either way, the error I get is the following:

Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 6, in <module>
    from motioneye.meyectl import main
  File "/usr/local/lib/python3.7/dist-packages/motioneye/meyectl.py", line 28, in <module>
    import settings
ModuleNotFoundError: No module named 'settings'

Does anyone have any idea as to what's causing this?

ualex73, 2019-07-29:
Python3 isn't support yet with MotionEye. You can use/try the Python3 port: https://github.com/ualex73/motioneye/tree/python3 (based on 0.40)

testbughub, 2019-07-29:
Thanks for the reply and link.
However, now I get another error:

Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 11, in <module>
    load_entry_point('motioneye==0.40rc5', 'console_scripts', 'meyectl')()
  File "/usr/local/lib/python3.7/dist-packages/motioneye-0.40rc5-py3.7.egg/motioneye/meyectl.py", line 256, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python3.7/dist-packages/motioneye-0.40rc5-py3.7.egg/motioneye/server.py", line 446, in main
    daemon.start()
  File "/usr/local/lib/python3.7/dist-packages/motioneye-0.40rc5-py3.7.egg/motioneye/server.py", line 111, in start
    self.daemonize()
  File "/usr/local/lib/python3.7/dist-packages/motioneye-0.40rc5-py3.7.egg/motioneye/server.py", line 72, in daemonize
    si = file('/dev/null', 'r')
NameError: name 'file' is not defined

ualex73, 2019-07-29:
Where did you get pypi motioneye==0.40rc5 version from? The pypi hosts a python3 version of motioneye, but that's really a none working python2 version (please remove it).

BTW - the "si = file('..." doesn't exist in my python3 port, so it is using the wrong one now.

testbughub, 2019-07-29:
I honestly don't know. I installed everything with apt and pip.
I'm gonna try to remove everything and start from scratch.

testbughub, 2019-07-29:
So I removed motioneye and motion, then installed motion and then the python3 motioneye and I get the same error as before.
How do I remove pypi?

testbughub, 2019-07-31:
I didn't get it working, so I flashed a new card with diet-pi Buster and chose to install motioneye with their cli software.
Now everything works, so I'll close this.

closed 2019-07-31 by testbughub