motioneye-project/motioneye #1391: Help getting a china 360 IP camera to work with motioneye

testbughub, 2019-07-31:
Hi!

I have this IP camera I bought from wish.
Everything works with the app I was instructed to use, but now I want it to work in motioneye.
The motor functions can be a later problem. I just want it to show me the video stream for now.

The info I get from the app about the model is:
PW201506A-GTY
However I couldn't find any google results for this device.

I did an nmap on it, and the only open ports on it is 23456 and 34567, which I find odd, but what do I know.

Any help would be appreciated!

testbughub, 2019-08-01:
Anyone got any ideas?

zagrim, 2019-08-01:
If you have no idea of the interface of the camera, it's pretty hard. But, if everything else fails, you could possibly use something like Wireshark to capture the traffic between the app you were instructed to use and the camera to see what URLs are being used. Search for HTTP/HTTPS connections using GET verb, some of the accessed URLs should probably be the video stream.

testbughub, 2019-08-01:
I've no experience with Wireshark, so I don't know if I'm doing this correctly,
but all I get from the camera is ARP requests.
Screenshot: https://imgur.com/bRu2TEU

zagrim, 2019-08-02:
Basically you should capture with a filter like `host <IP of the camera>`, which would capture everything to and from that IP address. If that gets too noisy with broadcasts etc, you could further limit it to the ports you found open with adding `and (tcp port 23456 or tcp port 34567)` to the capture filter, but in case the camera somehow masks some ports from the scan it has a chance of missing something.

Did you have the other camera software running on the same host and communicating with the camera when you ran the capture? There must be something else than just ARP requests if there's some communication ongoing.

testbughub, 2019-08-02:
I must have done something wrong, because I did use the app as I was capturing, but all I could see was ARP requests.
I tried filtering just the IP and port 23456/34567, but nothing comes up. 
It feels like I'm doing something wrong.
I really appreciate your help.

jasaw, 2019-08-05:
Use nmap to probe both TCP and UDP ports.
If you want to use wireshark, you'll need a managed switch to duplicate all traffic to your PC running wireshark, otherwise you won't see any traffic except for ARPs.

zagrim, 2019-08-05:
My bad for not clearly stating my of running the capture on the same host as running the camera app. Anyway, from the screenshot it looks like the camera is in wireless network, in which case I think every host gets anyway to see all the packets, but capturing needs to be run with ["monitor mode" enabled](https://wiki.wireshark.org/CaptureSetup/WLAN#Monitor_mode) (which does not on Windows, and I guess might not be available on all adapters).
So, since I'm guessing the app the camera vendor recommended or bundled is only for Windows (they usually do, sigh), it would be easiest if Wireshark was run on the same Windows host.

testbughub, 2019-08-05:
I'm about to give up on this idea.
I did a thorough scan on the ports and the services are: 
23456/tcp open  aequus?
34567/tcp open  dhanalakshmi?

Looks like gibberish to me, but maybe someone of you recognize it.

testbughub, 2019-08-05:
Also, I tried putting my wifi in monitor mode before sniffing, but I only got 1 ARP request when my phone connected to the camera.
I'm not sure if I'm doing this right, so if someone could do a detailed guide, I can see if I did it right or not.
And I must say thank you both for helping this little newbie.

zagrim, 2019-08-05:
@testbughub You mentioned phone. Could it be that this is one of those cameras that are wired to the cloud out-of-the-box, and the camera app in the phone fetches the stream from cloud. That would be a definite no-go regarding any locally installed software.

testbughub, 2019-08-05:
Yeah, I think it's cloud based.
I was afraid of that. So there's no way to intercept the video stream?

zagrim, 2019-08-05:
To intercept the stream? Sure by placing the camera behind a firewall and capturing the traffic. 
To actually be able to use the video stream? With some serious hacking, maybe, but I don't think it's worthwhile.

testbughub, 2019-08-05:
Yeah that's what I figured. Oh well..
But speaking of motor IP cameras, is there any that works with motioneye?
Or should I stick with the simple ones?

jasaw, 2019-08-05:
Use wireshark to capture all packets to & from this camera. Inspect the contents of the packets to figure out which protocol it is running. Hope that the comms is not encrypted. I imagine it probably use standard protocols like MQTT & RTSP. There may be some sort of authentication at the beginning too. Once you've reverse engineered the protocols, you can implement your own server on your PC.

You'll also need to force the camera to talk to your PC rather than the cloud server. Use wireshark to see which server address it is trying to resolve in its DNS request. Update the DNS entry for this server address to point to your local PC (something like dnsmasq). Now the camera should be attempting to establish connection to your local PC instead.

I think it's not worth your effort to do all of the above given that IP cameras are quite affordable.

testbughub, 2019-08-05:
Thanks @jasaw , but I'm gonna put this camera in the bin.
I could probably do what you said, but as you said at the ending, it's just not worth the trouble.

zagrim, 2019-08-06:
I don't have any experience myself, but there's some talk about IP cameras with pan-tilt-zoom capabilities in #122 (and a few others found by searching the issues with "ptz") with some brands and models mentioned, all probably outdated by now, though, but I get the idea that except the ones supporting ONVIF the motor control interface varies quite a bit and requires some work to make things work with the camera action buttons.

testbughub, 2019-08-06:
I see.
Well, this project is a fail.
But I thank you both for your input.

closed 2019-08-06 by testbughub