motioneye-project/motioneye #1392: usb&remote cameras no action on motion

5145h, 2019-07-31:
Hi, 

It seems that the motion triggering is not working on usb and remote cameras.
I can see motion on camera and even if I have all the configuration ok (from my point of view) the snapshots/movies are not saved on disk.

Thanks.

MobilisInMobilum, 2019-08-12:
Hello, something similar happened to me with rspberry cam on pi zero w.
Can you disable under "Motion detection" category following:
Auto noise detection -> Off
Noise level -> 0
Light Switch Detection -> 0

apply and check again?

norm42, 2019-11-22:
This fix worked for me with a Logitech C920 Pro HD Webcam on a usb port.  
Thx

JamBalaya56562, 2026-11-07:
Closing, as the settings above fixed this for another user: Auto Noise Detection off, Noise Level 0 and Light Switch Detection 0 under Motion Detection. Please open a new issue with your motionEye log if it still happens.

closed 2026-11-07 by JamBalaya56562