motioneye-project/motioneye #1393: Upgrade results in camera no longer being configured

keverets, 2019-08-01:
Just attempted to update on Ubuntu 18.04.1 using the documented instructions:

```
pip install motioneye --upgrade
systemctl restart motioneye
```

This was going from 0.39.4 to 0.40.0.

After upgrading, visiting the admin interface gives:

```
You have not configured any camera yet. Click here to add one...
```

That was unexpected and disappointing.

zagrim, 2019-08-02:
Take a look at the [motionEye release notes](https://github.com/ccrisan/motioneye/releases) for 0.40, you need to run a configuration migration script.

keverets, 2019-08-03:
OK. It would have been nice to have a warning or that autodetected at the time of upgrade. Buried in the bottom of a release note makes it unlikely to be seen.

zagrim, 2019-08-03:
I kinda agree with you that it would be user-friendly to have something in the upgrade process to remind the user to either check the release notes, or tell directly what to do. I just have no idea if it is possible to detect upgrade situation. Maybe better error handling in the application (loading of the configuration) could be used to inform the user, but it might as well prove to be ugly and hard to maintain over time.

digiltd, 2019-08-16:
This caught me as well, I couldn't update the main wiki (as I assume you want control) but a page under the `troubleshooting` section documenting the problem would be very helpful.

```
### Previously configured cameras not showing after upgrade

If upgrading from `3.x.x` to `4.x.x` your previously working cameras will not appear, with the following message in their place: `You have not configured any camera yet. Click here to add one...`

Starting with motionEye 0.40, the new Motion 4.x directives are used in configuration files. You can either remove and re-add all your cameras or you can use the ![migrateconf.sh](https://github.com/ccrisan/motioneye/blob/master/motioneye/scripts/migrateconf.sh) script to convert your configuration files:

`for file in /etc/motioneye/{motion,thread-*}.conf; do /usr/local/lib/python2.7/site-packages/motioneye/scripts/migrateconf.sh $file; done`

(depending on your install, you might need to replace `site-packages` with `dist-packages` and add `sudo` just after `do`)
```

Ultrabenosaurus, 2019-10-26:
@digiltd I assume I need to turn motionEye and motion off, run that script, then manually update motionEye and restart them both?

digiltd, 2019-10-27:
@Ultrabenosaurus umm, it was a while ago, but yes that sounds about right. Take backups of everything before running it though and be sure you are pointing to the right locations/files first.

JamBalaya56562, 2026-11-07:
Closing as answered: the 0.40 release notes describe this change, and `migrateconf.sh` converts the old camera configuration. You can also remove and re-add the cameras. The script still ships in `motioneye/scripts`.

closed 2026-11-07 by JamBalaya56562