motioneye-project/motioneye #1394: Motioneye camera gets frozen

guilhermetb, 2019-08-03:
Hello .. Good night.
Need help.
Bought 2 "gocomma CA - R21A - R Wireless 1080P HD Smart WiFi IP Camera - Black"
They are excellent cameras, when I use the software "ONVIF Device Manager" the live video option opens the address rtsp: //192.168.1.101: 554/11 and the picture appears perfect without any delay.
I added the camera in motioneye, as soon as I add the image of the camera but it is frozen giving a possible refresh after minutes and does not work the live mode
Can anyone help me how to solve this problem?
[image]

guilhermetb, 2019-08-03:
My raspberry is model 2 but img 2 does not work, it does not start.
I am using raspberry 1 image and it works correctly

I've used these versions and in both the camera image is frozen
motioneyeos-raspberrypi-20190619
motioneyeos-raspberrypi-20190427

I used the camera connected to wifi and cable and the image is still frozen.

zagrim, 2019-08-03:
What video resolution the cameras are producing? If it's Full HD 1080P it might just be too much for RPi2 processing power, especially if the image you're using doesn't provide support for hardware H.264 decoding.
You could `ssh` in the RPi and check the CPU load using `top`. I guess you ran that "ONVIF Device Manager" on something else than RPi?

guilhermetb, 2019-08-03:
Really the problem is the hardware
I did some research and concluded that my raspberry is version 1
The processor was constantly above 90%
I will buy an NVR, I believe it is a better cost x benefit

[root@motioneye ~]# uname -a
Linux motioneye 4.14.98 #1 Sun Apr 28 10:27:40 UTC 2019 armv6l GNU/Linux
[root@motioneye ~]# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 000e
Serial          : 000000004823b7d8

zagrim, 2019-08-04:
Dedicated HW is surely less time spent in maintenance and all kinds of fuss, especially if you intend to get more cameras. Myself I'm running 2 IP cams on RPi2B, and I need to do motion detection on a low quality substream due to the lacking CPU capacity, but thanks to the nice design of Motion (`netcam_highres` option) I can still save higher resolution movies with decent quality. But still, a RPi is not a perfect solution.

Please close this issue if it is now resolved.

guilhermetb, 2019-08-04:
Thank you for your help,
I love raspberry pi and this is a great project but to work properly i would have to buy a new board (pi 3) and its value ranges R$200~300
With the same value I can buy a NVR 4 channel IP Full HD with support 1TB+ storage

closed 2019-08-04 by guilhermetb