motioneye-project/motioneye #1395: Add username/password to already added rtsp camera stream

JBS5, 2019-08-04:
I have several rtsp streams added to motionEye. One of them without username and password. On the IP camera side, I want to add this authentication but I am wondering if it is possible to add this authentication to an already added camera in motionEye. Would be nice to keep the camera in motionEye without the need of re-add them because of all the setting including motion mask etc.
But I can not found fields for username and password when the camera is already added. Only when adding a new one.

zagrim, 2019-08-04:
You can add `netcam_userpass` Motion configuration option either via "Extra Motion Options" in the motionEye GUI, or if it somehow fails (I think there shouldn't be any technical reason for that, I've overridden motionEye defaults that way) edit directly the camera config file (`/etc/motioneye/camera-*.conf` if you have standalone up-to-date installation of motionEye). The format of the option is `netcam_userpass <username>:<password>`.

JBS5, 2019-08-04:
Thanks. Adding netcam_userpass admin:admin works like a charm.

closed 2019-08-04 by JBS5