motioneye-project/motioneye #1399: web interface doesn't load "No closing quotation"

dadosch, 2019-08-06:
Ok, so I'm a bit in a hurry as I'm on the site, but motioneye(os) stopped working after configuring motion notification via email (with special chars like ./#+ etc) and configuring motion mask.

The log says
```
2019-08-06 15:25:35: [motioneye]    ERROR: No closing quotation
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 273, in get
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 660, in list
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1382, in motion_camera_dict_to_ui
  File "usr/lib/python2.7/shlex.py", line 279, in split
  File "usr/lib/python2.7/shlex.py", line 269, in next
  File "usr/lib/python2.7/shlex.py", line 96, in get_token
  File "usr/lib/python2.7/shlex.py", line 172, in read_token
ValueError: No closing quotation
```

dadosch, 2019-08-06:
`'` is probably the special char which is causing the error

zagrim, 2019-08-07:
I guess everything passed to `shlex.split()` should be validated against (shell) special chars (at least unescaped ones) so it doesn't choke on them.

Since you were able to pinpoint the setting that caused the issue, I guess you got this worked around and got the application working again?

CRImier, 2019-08-07:
You can probably use `shlex.quote()` to make the password (and login) shell-safe?

dadosch, 2019-08-07:
> I guess everything passed to `shlex.split()` should be validated against (shell) special chars (at least unescaped ones) so it doesn't choke on them.
> 
> Since you were able to pinpoint the setting that caused the issue, I guess you got this worked around and got the application working again?

Yes, thanks for asking. I had to use a different mail account with no `'` in the password, since changing this mail password would break many things on my server (noreply account)

Interestingly enough, the test mail got send out successfully (with `'` in the password) if I clicked on the button.

free-cup-holder-exe, 2026-10-04:
This still happens on current `dev`. A `'` in the SMTP password, or in any other email or Telegram notification setting, is written into the single-quoted `sendmail`/`sendtelegram` command unescaped. `shlex.split()` in `motion_camera_dict_to_ui()` then fails with "No closing quotation", and `/config/list` and `/config/<id>/get` return HTTP 500, so the UI and API clients such as Home Assistant show no cameras at all, not only the affected one. The notification doesn't work either, because `sh` can't parse that `on_event_start` line.

#3434 stops the crash by showing a part that can't be parsed as a custom command. #3426 removes the cause, because these settings are then kept in the camera config instead of on the command line.

Until then, removing the `sendmail` or `sendtelegram` part from `on_event_start` in `camera-<id>.conf` and restarting motionEye brings the cameras back, and the notification can be set up again with a password without `'`.

MichaIng, 2026-10-05:
Single quote characters are now allowed in email and Telegram passwords, as they are stored in the camera config, instead of being passed as `on_event_start` command argument. This is live in `dev` branch, and will be part of the v0.45.0 pre-releases and release.

closed 2026-10-05 by MichaIng