motioneye-project/motioneye #1400: USB Webcam not supported?

bforpc, 2019-08-07:
Hi,
the usb  webcam is detected correctly and lsusb shows this (the first one is the cam):
```
Bus 001 Device 005: ID 0c45:6366 Microdia
Bus 001 Device 004: ID 046d:c312 Logitech, Inc. DeLuxe 250 Keyboard
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

ls /dev/video*
/dev/video0  /dev/video1  /dev/video10    /dev/video11  /dev/video12

I can -if motioneye ist off - make pictures from the cam, with fswebcam correctly.

This is the syslog when starting motioneye:
```
Aug  7 08:48:42 schuppen meyectl[638]:     INFO: hello! this is motionEye server 0.40
Aug  7 08:48:46 schuppen motion: [16303752:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Aug  7 08:48:46 schuppen motion: [16303752:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Aug  7 08:48:46 schuppen motion: [16303752:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "vid_control_params"
Aug  7 08:48:46 schuppen motion: [16303752:motion] [NTC] [ALL] motion_startup: Motion 4.1.1 Started
Aug  7 08:48:46 schuppen motion: [16303752:motion] [NTC] [ALL] motion_startup: Logging to syslog
Aug  7 08:48:46 schuppen motion: [16303752:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Aug  7 08:48:46 schuppen motion: [16303752:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Aug  7 08:48:46 schuppen rsyslogd: action 'action-6-builtin:omfile' resumed (module 'builtin:omfile') [v8.1901.0 try https://www.rsyslog.com/e/2359 ]
Aug  7 08:48:47 schuppen motion: [1:ml1:Camera1] [ERR] [VID] v4l2_set_mmap: Error starting stream. VIDIOC_STREAMON: Protocol error
Aug  7 08:48:48 schuppen meyectl[638]:     INFO: cleanup started
Aug  7 08:48:48 schuppen meyectl[638]:     INFO: wsswitch started
Aug  7 08:48:48 schuppen meyectl[638]:     INFO: tasks started
Aug  7 08:48:48 schuppen meyectl[638]:     INFO: mjpg client garbage collector started
Aug  7 08:48:48 schuppen meyectl[638]:     INFO: server started
Aug  7 08:48:57 schuppen motion: [1:ml1:Camera1] [ERR] [ALL] motion_init: Error capturing first image
Aug  7 08:48:57 schuppen motion: [1:ml1:Camera1] [ERR] [ALL] mlp_capture: Video device fatal error - Closing video device
Aug  7 08:49:00 schuppen motion: [1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Aug  7 08:49:00 schuppen motion: [1:ml1:Camera1] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Aug  7 08:49:00 schuppen motion: [1:ml1:Camera1] [ERR] [ALL] mlp_capture: Video device fatal error - Closing video device
```
Whats wrong here?

Bfo

CRImier, 2019-08-07:
Hi! Any chance it's a cheap webcam? These often have incompatibilities with i.e. Linux. It's weird that you get 5 devices, too - there only needs to be one. Does `fswebcam` work with only one of these devices, or all of them?

jasaw, 2019-08-08:
@bforpc Some `/dev/video*` device implements hardware accelerated video/image compressor, some implements camera (video/image capture).
To find out what each /dev/video* implements, run `for d in /dev/video* ; do echo $d ; v4l2-ctl --device=$d -D --list-formats  ; echo '===============' ; done`
You can replace `--list-formats` with `--all` to give you more information about a particular v4l2 interface.

bforpc, 2019-08-14:
Hi,
@CRImier : No it was a very well usb webcam. fswebcam has worked with all devices
@jasaw : i have send the webcam back and use now a ip based cam, what is working very well with motioneye. So the Problem is solved, but in a other matter ;-)
Thanks all of you.
bfo

JamBalaya56562, 2026-11-07:
Closing, as you sent the webcam back and an IP camera works well with motionEye for you. The cause with the webcam stayed unknown, so anyone seeing the same can open a new issue with the output of `v4l2-ctl`.

closed 2026-11-07 by JamBalaya56562