motioneye-project/motioneye #1401: MotionEye - does it run on Chrome 76.0.3809.100?

potto, 2019-08-07:
Hi, can you please check if MotionEye does work on Chrome 76.0.3809.100 64bit slovak language? I have no stream from cameras. It works normally on Edge. Thank you

AZRadiohead, 2019-08-07:
I was just in the process of posting about this.  I'm on Google Chrome for Android 76.0.3809.89 and it stopped working.  I disabled it to revert to an older version and it worked.  As soon as it updates it breaks the stream view.  Another family members phone was working till I updated it to the newest chrome and broke theirs.  They aren't happy with me. 
So, @ccrisan do you have any wisdom to share with is since it appears that Google changed/broke something in the latest versions of Chrome?  I also tried FireFox and it works fine there.
Thanks

breakingspell, 2019-08-07:
Chiming in as well, observed as an update from Chrome 75 to 76 reproduced this behavior. I'm seeing a spinning indicator with no errors in Chrome's console. Other browsers work fine, as reported. 
I only use a single webcam, so I spent a while checking the cam and nginx reverse proxy config before checking Github for issues and confirming Chrome as the common factor. 

Motioneye's logs don't show any errors, while Nginx's error logs show this interesting snippet, minus identifying info (I use motioneye under https://host/cam/). Let me know if there's any other logs we can gather to identify what might be cutting the stream up. 

```
2019/08/07 14:48:49 [error] 16619#16619: *9205 upstream prematurely closed connection while reading response header from upstream, client: 192.168.1.1, server: redacted, request: "GET /cam/picture/1/current/?_=1
565207319188&_username=&_signature= HTTP/1.1", upstream: "http://internalIP:8765/picture/1/current/?_=1565207319188&_username=&_signature=", host: "redacted", referrer: "https://redacted/cam/"
```

ccrisan, 2019-08-07:
The fix for this issue is already on motionEye dev branch: https://github.com/ccrisan/motioneye/commit/b22ef62dfa02dc5b09c84fca557f561043eb6f78.

The build server is temporarily down, but I'll try to put together a new nightly build throughout this week.

breakingspell, 2019-08-07:
Confirmed that the changes in the commit did the trick. Thanks!

woznyp, 2019-08-08:
@ccrisan any progress in the hotfix? (as I understand after the hotfix we can update automatically the os and web interface using Software Update Check, am I right?)

grhangelone, 2019-08-08:
I'm not sure if that's all we have to do.
I now see my previews again, but a lot of things are broken.

- Fullscreen doesn't work
- The configuration pane on the left is completely grayed out
- 2 of 4 cameras show the same preview picture but the picture browser for them works

[image]

woznyp, 2019-08-08:
@grhangelone did you reboot Raspberry after changing JavaScript files? Also clear browser cache and history.
This is what I did and now I see streams and I can configure any of mine cameras (native and usb)

grhangelone, 2019-08-08:
I did clear the cache and also tried another browser.
Opera does the same.
So i might have another problem here.

AZRadiohead, 2019-08-08:
After looking at what was fixed in the dev branch I found a temporary solution to fix it in Chrome on the Desktop (PC in my case) and on my Android Chrome.
In your Chrome browser you can go to chrome://flags/#enable-lazy-image-loading and change the option from Default to Disable and reload Chrome.  The cameras worked again after I disabled 'Lazy Frame Image Loading'.  There is also another flag, chrome://flags/#enable-lazy-frame-loading but I did not need to change it for mine to work.
Not knowing all the ramification of this it's probably best to reset these once you have the latest fix for Motioneye.  I'm sure the developer will chime in if there's anything additional we should be aware of.

huzhiren, 2019-08-10:
Can be worked around for now by disabling chrome://flags/#enable-lazy-image-loading

bob-lee, 2019-08-10:
@ccrisan still have those code changes ([b22ef62](https://github.com/ccrisan/motioneye/commit/b22ef62dfa02dc5b09c84fca557f561043eb6f78)) on my `rpi zero w` and can confirm that they work ok with default flag on:
- Chrome 76 Windows 64
- IE11 Windows 64
- Chrome 76 on ios
- Safari on ios

ccrisan, 2019-08-11:
I'll make sure to issue a new release today. I'll have to take care of motionEyeOS as well.

ccrisan, 2019-08-11:
Version 0.41rc1 has now been published. Please install it, test it and let us know if anything should be adjusted or fixed before 0.41 stable.

closed 2019-08-11 by ccrisan

smokeybandit, 2019-08-12:
0.41rc1 fixes the issue for me.  Thanks.

Prutsium, 2019-08-19:
Same here running Motion 4.2.2 and Motioneye 0.41rc1 on Ubuntu with the rc1 update all loading issues fixed.

jonney2069, 2019-08-21:
Hello, I've been having this problem as well.  Can anyone point me in the correct direction for upgrading motioneye? Thank you!

potto, 2019-08-21:
@jonney2069 You can enable prereleases from your Settings menu (Expert Settings). Then update as usual.

transmitamit, 2019-08-24:
Post update to pre-release version, wpa_supplicant fails and keeps rebooting on raspberry pi 3 B+. Anyone else facing the issue.?

2E0PGS, 2019-08-29:
Glad I came across this. Had me scratching my head for a while. Given chrome must of auto updated it self...

glyderman8, 2020-04-10:
Hi, I have just found out today, 10-04-20, that my motioneye viewing on raspian stretch was not working on two camera I use it on, when I tried to view the stream on Firefox. Thinking something had changed I updated raspian and updated motioneye to latest version. It made no difference. I then decided to use my phone. It worked ok on both cameras. One uses a webcam and other uses the raspberrypi camera. So I then realised it was probably firefox. I then used a laptop I hadn't used for a while and firefox worked ok. It was version 74.01. I let it update to version 75 and tried again. It then did not see the stream from motion eye. So latest firefox has broken the viewing capablity from moitoneye. I noticed a workaround for chrome and tried similar with firefox. That is: about:config and change the lazy-image flag to false. Guess what... it worked. So for now this workaround works.

2E0PGS, 2020-04-13:
Mine works fine now after updating motion-eye.