motioneye-project/motioneye #1405: Camera suddenly not recognising on any of three motioneye pi's

beefpi, 2019-08-09:
Yes that's right, all three motioneye pi's across two different networks have stopped showing the view from the camera in the interface, and I just get the rotating graphic infinitely.
I did a check for update and realised I was a few versions behind so have updated to 20190619, and the problem persisted.
I took no chances and formatted my SD card and installed 20190619 again but from scratch and still the problem persists.
I've googled "motioneye camera not working" and found a few people with cameras not working but most of the posts are over a year old, and therefore must have been several versions behind. Also, most were solved with a reseating of the camera on the board, but as I have not touched either of my three boards, I cant imagine that is either the problem or the solution.
Any thoughts would be appreciated.
[image]

dieselrabbit, 2019-08-09:
This seems to be the same as https://github.com/ccrisan/motioneye/issues/1401

ragecc, 2019-08-09:
I woke up this morning to no cameras showing. I thought maybe the sd card got corrupted or something so I formatted and reinstalled. Still not showing. After some digging I found out it's most likely a issue with chrome so I opened up Microsoft Edge for the first time this year and typed in the motioneye ip address. To my surprise the camera showed. I don't know what we do to get a fix to use chrome again. Maybe somebody will chime in on that part.

Lefuneste83, 2019-08-11:
The issue is indeed related to Chrome update... There is a workaround by disabling a setting in Chrome...

https://support.google.com/chrome/thread/11506284?hl=en

beefpi, 2019-08-11:
You were spot on.
Changed the setting and problem solved.
Many thanks for this.
Pretty peeved at chrome because I rebuilt all three of my motioneye raspberries thinking they needed an update when i needn't have done anything! :-(
At least I'm up to date anyway.
Thanks again. :-)

closed 2019-08-11 by beefpi