motioneye-project/motioneye #1406: Change Thumb Filename

postal302, 2019-08-10:
I changed the filename method for recorded videos to include the date. Now the created thumb images will not show when previewing the recorded movies. Is it possible to change the name of the thumbnail files or have the thumbnail files match the naming convention of the video?

I tried changing the still image filename but this did not change the thumbnail image name.

zagrim, 2020-07-04:
It's been a while since you created this issue, but I just tried to reproduce it and couldn't find any issue. When you said "I changed the filename method for recorded videos to include the date" do you mean adding the date part to the video file name as, for example "%Y-%m-%d" as a prefix, so that the whole content of "Movie File Name" could be e.g. "%Y-%m-%d/%Y-%m-%d_%H-%M-%S"? For me, with the most recent developer build, this works just ok. It even works if I remove the folder part completely, so that all files would go directly under the directory named after the camera (e.g. "Movie File Name" -> "%Y-%m-%d_%H-%M-%S").

postal302, 2020-07-09:
I concur, this is resolved. I input the same structure for the image and video filenames and it is working correctly now.

Thank you to the folks who've built a great thing here! This issue is closed.

closed 2020-07-09 by postal302