motioneye-project/motioneye #1408: MotionEye does not work with RasPi Cam

snoopy27, 2019-08-10:
Pi 3B+, RPi Cam v2
"Raspbian GNU/Linux 10 (buster)"
motion Version 4.1.1
# pip show motioneye
Name: motioneye
Version: 0.40
# v4l2-ctl --list-devices
bcm2835-codec (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
mmal service 16.1 (platform:bcm2835-v4l2):
        /dev/video0

http://192.168.0.71:8081  is showing the pictures from motion (low level),
but with motioneye under http://192.168.0.71:8765 the camera box with the circling in the middle remains empty (with mmal as well with V4L2/bcm2835-codec).

wintdkyo, 2019-08-12:
Having the same issue with:
Pi Zero W, Raspbian 9.8
motionEye Version | 0.40
Motion Version | 4.1.1

The streaming page (port 8081) shows the video stream but not on under the admin page under port 8765. Getting a circling icon, and then a gray screen in the admin page.

CRImier, 2019-08-13:
Hi! Any chance any of you (both @snoopy27 and @wintdkyo ) are using Google Chrome? It recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

wintdkyo, 2019-08-13:
Thank you @CRImier ! For me Chrome seemed to be the issue. Appreciate it!

JamBalaya56562, 2026-10-30:
This was caused by a change in Chrome 76 that broke the camera streaming in the web UI, see #1401. It was fixed in motionEye 0.41 (b22ef62), and the second reporter confirmed that Chrome was the cause. Closing this; if the camera still does not show in a current motionEye release, please open a new issue with your details.

closed 2026-10-30 by JamBalaya56562