motioneye-project/motioneye #1411: Motion eye OS not recognizing my USB camera

mish11mish11, 2019-08-12:
The web UI shows a loading circle that goes on forever and when trying to take a still though the command line I get this.
mmal: Cannot read camera info, keeping the defaults for OV5647
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:EN                                                                                        OMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is instal                                                                                        led correctly

CRImier, 2019-08-12:
Are you sure you're trying to enable a USB camera? These messages look like the ones you'd get when connecting to a Raspberry Pi Camera. In addition to that, you might be affected by the recent bug that prevented MotionEye from working with Chrome - unless you got the latest revision there is (IIRC it was released in the last 24 hours), try Firefox.

mish11mish11, 2019-08-12:
Wow, thank you so much it was just simple as using edge instead of chrome. Do you think this bug is going to be patched soon?

CRImier, 2019-08-12:
It's already patched, [use this release](https://github.com/ccrisan/motioneye/releases/tag/0.41rc1). Also, the output you posted is not related to the Chrome issue, so, you might want to check if your cameras are set up right =)

mish11mish11, 2019-08-12:
yeah, I done goofed I was trying to take a still from a nonexistent pi camera instead of USB :)

mish11mish11, 2019-08-12:
Also, I don't know if you know anything about this but I have this old Xfinity home security network camera do you know if it's possible to connect that to motion eye.

CRImier, 2019-08-12:
You will need to do some research on your own - power it up, connect to it and see if it outputs i.e. a MJPG stream. I'm not a MotionEye dev, but I imagine there's 1) documentation for setting up IP cameras with MotionEye 2) some related issues in this repo that you can search through =)

mish11mish11, 2019-08-12:
Yup, I'm definitely going to do some research just thought id ask if you had prior experience with those type of cameras. Regardless thanks for your help.

CRImier, 2019-08-13:
Happy to help! Please close the issue =)

closed 2019-08-13 by mish11mish11