motioneye-project/motioneye #1412: Spinning icons instead of camera feed

mattpstanding, 2019-08-13:
Out of the blue both of my camera feeds (one local and one remote RPI camera) have stopped working. When opening the webpage, the feed is not displayed and instead I get perpetual spinning wheels.

Both cameras are are recording motion and image/video files.

The master camera is a on a Pi3 running:
MotionEye: 0.40
Motion: 4.1.1
OS: Raspbian 9.9

The second camera is a Pi Nano W and is running the same version but using the MotionEyeOS image.

The setup has been running perfectly for about 12 months without issue but both cameras developed this issue overnight a couple of days ago.

CRImier, 2019-08-13:
Hi! It seems like you're using Google Chrome, it recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

bob-lee, 2019-08-13:
Yesterday I upgraded motioneye and it seemed working ok on chrome:
sudo pip uninstall motioneye
sudo pip install motioneye==0.41rc1

closed 2019-08-14 by ccrisan