motioneye-project/motioneye #1413: No "advanced settings" button

tom120934, 2019-08-13:
I'm using 0.41RC1 image in Docker (ccrisan/motioneye:master-amd64).

In General Settings, there is no "Advanced settings" button : the first line is "Admin Username".

I've checked the whole settings panel, no "advanced settings" elsewhere.

Thanks for reading, thanks for any help !

zagrim, 2019-08-15:
What "advanced" option are you missing, then?

tom120934, 2019-08-16:
Thanks for the reply. I was looking for the timezone / datetime settings, read they're in the "advanced settings".

zagrim, 2019-08-16:
I'm not quite sure of this (I've never run dockerized motionEye or motionEyeOS), but I've not seen "Advanced Settings" in the user interface (there has been such thing, that's for sure). However, there's something in the [Docker instructions](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker) on how to set the timezone of the container, and at least my installation of non-dockerized ME has the following in `/etc/motioneye/motioneye.conf`, too:

```
# path to the localtime file
# (enable this to configure the system time zone from the UI)
#local_time_file /etc/localtime
```

tom120934, 2019-08-19:
Thanks, I solved my timezone issue that way. But I still don't understand if the "advanced settings" exist in Docker.

ccrisan, 2019-08-19:
Advanced settings filter is gone for good. All available settings are now shown. If timezone is not there, make sure you have set the [local_time_file](https://github.com/ccrisan/motioneye/wiki/Configuration-File#local_time_file) correctly in your settings.

closed 2019-08-19 by ccrisan

tom120934, 2019-08-20:
OK, good to know. Thanks for the good work !