motioneye-project/motioneye #1414: No camera image when a special character is inserted in text overlay section (custom text)

lukasig, 2019-08-13:
Hi

I have noticed that when inserting a special character (in this case the "â" letter from the romanian alphabet) into the custom text entry of the text overlay section, the camera image or stream does not come on anymore.

It is not anything of much importance, but I thought I should mention it.

zagrim, 2020-07-04:
I can't reproduce this (with a build of the head of dev branch), which is a good thing I guess. The non-ascii characters in custom overlay text are not shown, though, which is probably due to the font missing non-ascii glyphs. Anyway, this seems to work somewhat better now, and the remaining issue could perhaps be fixed by using an Unicode-compatible font.

JamBalaya56562, 2026-11-10:
Closing, as zagrim could not reproduce the missing picture with a later build; only the non-ASCII letters were left out of the overlay, which Motion draws with its own font. Please open a new issue if a custom text still stops the picture with a current release.

closed 2026-11-10 by JamBalaya56562