motioneye-project/motioneye #1415: Camera Works Outside MontionEye - MotionEye doesn't add it

jeffdickmanhpe, 2019-08-16:
Potentially related to:  https://github.com/ccrisan/motioneyeos/issues/2049
OS:  Raspbian 10
motionEye Version:  0.40
Motion Version:  4.2.2
Camera: Longruner IR-Cut

Camera is setup `raspistill -t 2000 -o image.png -e png` creates image on desktop to show camera is working.

When I add the camera to MotionEye as` Local V4L2` with `mmal service 16.1` I get the spinning wheel.  

[v4l-ctl-output.txt](https://github.com/ccrisan/motioneye/files/3510126/v4l-ctl-output.txt)

Once the camera is added, if I run raspistill again, I get the following error:

> pi@cameractl:/dev $ raspistill -t 2000 -o image.png -e png
> mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
> mmal: camera component couldn't be enabled
> mmal: main: Failed to create camera component
> mmal: Failed to run camera app. Please check for firmware updates

[camera-1.conf](https://github.com/ccrisan/motioneye/files/3510138/camera-1.txt)

Removal of camera from MotionEye results in raspistill working again.

No errors in motion.log

CRImier, 2019-08-16:
Hi! Any chance you're using Google Chrome? It recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

jeffdickmanhpe, 2019-08-16:
Dangit... yes, using Chrome.  Just switched to Safari, it works.  

Thank you very much for your help!

closed 2019-08-16 by jeffdickmanhpe