motioneye-project/motioneye #1419: Camera not showing in :8765

fatjuice, 2019-08-18:
I never had issues with this, perhaps I'm missing something.

I cannot see a local mmal camera in the :8765, but I can see it in the streaming port :8081.

What causes this?

I did the install the same as I always do.

did this with 2 raspberry pi 3, same issue.

CRImier, 2019-08-18:
Hi! Any chance you're using Google Chrome? It recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

fatjuice, 2019-08-19:
Hi, I am using chrome.  Thanks for the info, once I get home, I'll give it a try and give feedback.

Thanks!

mjarva, 2019-08-28:
@CRImier Absolute gem! I have been fighting with setup, doing reinstalls and lots of testing, and it was as simple as that. Firefox shows feeds neatly for both USB camera and Raspberrys own camera module. Chrome not so much.

aredon, 2019-09-02:
I also had this issue in Chrome but I recreated the camera and now it's fine?

JamBalaya56562, 2026-11-07:
Closing as a duplicate of #1401: Chrome 76 broke the way motionEye loaded the live pictures, which b22ef62 fixed in 0.41. As noted above, updating motionEye or using another browser solves it.

closed 2026-11-07 by JamBalaya56562