motioneye-project/motioneye #1420: Doesn't work with slow camera's.

TimLeijten, 2019-08-19:
I would like to use motioneye to monitor a few webcams around my house, but I would also like to monitor 2 public high way webcams. My own cameras work fine, but the high way cam's don't. They are the kind that only refreshes once every 10 seconds, and motioneye seems to not like this. (it works when I add it, but after 5 minutes it will permanently give me the camera with cross image).
I added it as a network camera and it automatically detected the type.

Motioneye does support them as it worked for a bit, but then it didn't.

So does motioneye have like an fps threshold or something that prevents the highway cams from being used?

How can I fix the issue?

Thanks!

TimLeijten, 2019-08-22:
Does anyone have any idea? My 0.1fps highway cams work fine on raw motion and configuring it there. So it's a motioneye specific issue.

cook32, 2019-08-29:
Did you increase framerate ? Do you have a good network bandwith ? Do you use Chrome ?

jasaw, 2019-08-29:
This could be a motion software issue. Please try **without** motioneye. Use this as your guide: https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue

JamBalaya56562, 2026-11-08:
Closing, as this was with the motionEye and Motion of 2019, and there were no further replies. You found that the same 0.1 fps cameras work with plain Motion, so the difference lies in the settings motionEye writes; comparing that camera's config file with your working Motion configuration would show it. Please open a new issue with both if a slow camera still drops out with current releases.

closed 2026-11-08 by JamBalaya56562