motioneye-project/motioneye #1421: Unable to open video device

matrix866, 2019-08-19:
Goodmorning. I had the problem with chrome and latest stable version of motioneye on my raspbian pi3. So I updated to the latest nightly. After the update the configuration was empty so I loaded an old backup but i didn't have the password of admin so I reset the password by editing the file /etc/motioneye/motion.conf. Success!
After this I am not able to see anymore the board camera of raspberry. I attached the motionlog which seems to have errors. I also tried deleting the /etc/motion/motion.conf because I think I could have corrupted it opening it by nano editor but I don't know how to restore it back. I did again all the motioneye installation procedure but it didn't went back: I only have the /motioneye/motion.conf but not the /motion/motion.conf. 
In these step I also got the "unable to open video device".
Any help for the 2 problem? (1) restore /etc/motion/motion.conf (2) fix unable to open video device.
Thank you.

[image]

CRImier, 2019-08-19:
Did you run [the config migrate script](https://github.com/ccrisan/motioneye/releases/tag/0.40) after importing the old config?

matrix866, 2019-08-19:
No. What If I would like to reset motioneye? I deleted anyway alla cameras from web editor.

CRImier, 2019-08-19:
After deleting the cameras, can you re-add them?

matrix866, 2019-08-19:
Ok problem fixed: the problem was the file /etc/motioneye/motion.conf with multiple camera-1.conf and thread-1.conf etc. I deleted all and restarted raspberry and it started again to work.

Thank you CRImier. For sure the problem was what you pointed out in the start. But I missed up the configuration file trying to fix the problem.

CRImier, 2019-08-19:
happy to help =) please close the issue!

closed 2019-08-19 by matrix866