motioneye-project/motioneye #1422: Motioneye captures photos and video to disk but not visible in UI

jcherndon, 2019-08-20:
I have 4 amcrest cameras that worked up to the their lastest firmware update about a week ago.
Their realtime feeds have stopped showing the the motioneye gui. I know motioneye is connected to correct RTSP URLs because I can capture photo/video to file. I have debug running, there is no errors in logs, I can see received events for the photos/videos to be written.
Photos/videos can be pulled up in UI viewer.

domsamal, 2019-08-21:
I have experienced an issue recently where I can no longer view the live stream from the home screen of the GUI - though everything else functions correctly including using the camera's streaming URL:portnumber. 

I initially thought it was attributed to a recent router fw upgrade which I have since discarded as the root cause. I believe there is an incompatibility with chrome browser Version 76.0.3809.100 (Official Build) (64-bit) (or potentially an installed extension).
extension). 

I have different versions of the MotioneyeOS running on different RPis yet the homepage UI does not display the live feed using Chrome. Using Firefox and MS Edge I can see the full UI as I have come to expect on all installations.

Perhaps you can try accessing the UI via Firefox or MS Edge and see if the live stream is there as it should be ?

jcherndon, 2019-08-22:
Yea, I tried another browser and the video shows up correctly. Guess this is a Chrome  76.0.3809.100 64bit issue .

closed 2019-08-22 by jcherndon