motioneye-project/motioneye #1424: Setting AWB for local Raspberry Pi camera

Pi-DD, 2019-08-21:
I am trying to pass the new AWB setting “greyworld” (for NOIR cameras) to Motion using ...
mmalcam_control_params -awb greyworld 
via the “Extra Motion Options” box. 

It does not seem to work. I then tried some other parameters listed in the Motion config documentation as examples (e.g. -hf)  and they do not seem to work either. 

The camera is shown as vc.ril.camera, MMAL camera so my understanding is that the parameters should work. 

Am I missing some step? Do I need to stop/restart Motion/MotionEye? Reboot? 

Is there anywhere I should check to see if this has modified a config file? motion.conf does not seem to change, nor does the conf file for the camera. 

Thanks.

jasaw, 2019-08-22:
Check `/data/etc/camera-1.conf` to make sure the extra motion options made it there.
Not sure if this PR has anything to do with it. https://github.com/Motion-Project/motion/pull/993

Pi-DD, 2019-08-22:
I’ve checked /etc/motioneye/camera-1.conf

The parameters I put into the web interface appear there and change as I change them. motion.conf in the same directory lists camera-1.conf so I assume it is picking up the parameters to pass to the Pi camera software. I have started/stopped both camera from web control page AND MotionEye via systemctl so as to make sure parameters are passed as I assume this must happen at start up - as far as I know you cannot change Raspivid parameters “in flight”. 

I’ve tried changing the parameters and it makes no difference. It looks (but I may be wrong!) as if it (motion?) always uses -awb auto no matter what you set. 

At the moment you can swap between the old and new AWB algorithm by using 
vcdbg set awb_mode 1 (new) and 0 (old) whilst MotionEye is stopped.

The “new” version “auto” creates a purple haze for NOIR cameras, hence new “greyworld” setting. 

Testing this via raspivid I can see that in “new” awb mode awb auto gives purple haze, awb greyworld works to give grey tone with ir only and “washed out” colour with normal lighting, as old “auto” used to do for NOIR. 

Since changing awb for Raspivid when awb mode is set to 1 changes the images as one would expect and makes no difference when you make the same awb changes via MotionEye web control I have to conclude that somehow the motion parameter changes are never reaching the Raspberry Pi camera software.

Pi-DD, 2019-08-22:
Closed in error!

Pi-DD, 2019-08-24:
It seems this is Buster related. Passing mmalcam parameters works on Stretch (except the new greyworld, which defaults to auto. The awb_mode workaround sorts this out by forcing the “old” algorithm and eliminating the “purple haze”). 
No parameters at all are passed on Buster,  it the AWB mode workaround is still needed.

Pi-DD, 2019-08-27:
As ever - user error! 
I finally noticed the documentation note that installation via apt-get does not configure the mmal camera (even though it was picked up as such by the Motioneye interface I use). On my Buster machines I had done it this way. 
Installed 4.2.2 by get/dpkg and parameters now pass, although "greyworld" does not seem to be there yet.

Pi-DD, 2019-08-27:
Closed, not a Motioneye issue.

closed 2019-08-27 by Pi-DD