motioneye-project/motioneye #1427: Unable to open video devices

girardpeter, 2019-08-23:
please help. 
showing unable to open video devices.
I alreqdy using the latest version.
i am using logitech cam c310

MajorGlory78, 2019-08-24:
Goodmorning,

I had a similar issue a few days ago with a new Microsoft Lifecam HD. For me the solution was selecting a different codec! When you add a camera, you can select a code. Most of the time MotionEyeOs will pick the right one. I had to change it manually for one camera. Try them all, you might get lucky.

Good luck!

JamBalaya56562, 2026-11-08:
Closing, as this got no further replies. "Unable to open video device" means Motion could not read the camera; for a USB webcam like the C310, choosing a different camera format (codec) when adding it, as suggested above, often helps. If it still shows with current releases, please open a new issue with `motion.log` and the output of `v4l2-ctl --list-formats-ext`.

closed 2026-11-08 by JamBalaya56562