motioneye-project/motioneye #1428: Video stream not visible in main UI

rvdeijk, 2019-08-25:
For some reason I'm unable to see my video streams in the main UI of motionEye..
When I grab the stream and open it for example in VLC it works fine.
The console mentions the following message: Unchecked runtime.lastError: The message port closed before a response was received.

What am I doing wrong? I already reinstalled motionEyeOS. I tested it with version 20190427 and version 20190619. In the past it worked fine. So I have no idea what have been changed..

rvdeijk, 2019-08-25:
After reading some other issues it seems to be related to Chrome 76. When I disable this setting in chrome: chrome://flags/#enable-lazy-image-loading everything works fine. 

When will there be a new release, I don't like to use nightly's.

bob-lee, 2019-08-26:
`0.41rc1` seemed working ok on chrome:
```
sudo pip uninstall motioneye
sudo pip install motioneye==0.41rc1
```

closed 2019-08-26 by ccrisan

virtuallight, 2019-09-03:
> After reading some other issues it seems to be related to Chrome 76. When I disable this setting in chrome: chrome://flags/#enable-lazy-image-loading everything works fine.
> 
> When will there be a new release, I don't like to use nightly's.

@rvdeijk Thanks for sharing. That fix worked for me as well 👍

pentiumduck, 2020-02-18:
> After reading some other issues it seems to be related to Chrome 76. When I disable this setting in chrome: chrome://flags/#enable-lazy-image-loading everything works fine.

Just perfect. Thanks a lot !!