motioneye-project/motioneye #1430: Camera resolution issue

Saku241, 2019-08-27:
I got a 1080p camera that works on my computer but on my Pi I can go max to 800x600

It works in 1080p if I put `v4l2_palette=8` in extra motion settings

Can anyone know this issue ? Is this a way to fix it ?

Saku241, 2019-08-27:
Also on Chrome it shows me this 
[image]

instead of the streaming camera (it works on firefox and edge)

Saku241, 2019-08-27:
Update : When I'm using v4l2_palette=8, the streaming is perfect but when it detects movement and start writting, my Pi crash from overheating

zagrim, 2019-08-27:
Regarding the streaming issue: Chrome recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

brietman, 2024-11-26:
I can confirm this with my Aukey pc-lm1e webcam. Because there is no wider image and 1080 has more interpolation, 720p is fine for my at this moment.

JamBalaya56562, 2026-11-15:
The camera delivers 1080p only in MJPEG, and `v4l2_palette 8` is what selects MJPEG, so that setting is the fix rather than a workaround. With motion 4.4 and newer the option became `video_params palette=8`, and motionEye dropped it again whenever the camera settings were saved. #3551 keeps it, so from v0.45.0 an Extra Motion Options entry `v4l2_palette 8` is kept, shown as `video_params palette=8`. The Chrome part was the change discussed in #1401. Closing; please open a new issue if it still fails on 0.45.

closed 2026-11-15 by JamBalaya56562