motioneye-project/motioneye #1434: motioneye wont start

el97, 2019-08-28:
Hi! I have been using motioneye for a while now. A couple of weeks ago i noticed that my pi crashed and i had to format the SD-card. unfortunately i got quite ill and forgot about this until now. I have installed a fresh version of raspbian and followed the instructions for installing motioneye. Now when i try to start it i got some strange errors on systemctl. I'm more of a follow instructions kind of person so I would like to have someone to help me on this one. Or at least help me with where i should start.

```
 motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-08-29 01:31:37 CEST; 1s ago
  Process: 6950 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 6950 (code=exited, status=1/FAILURE)

Aug 29 01:31:37 motioneye meyectl[6950]:     import motionctl
Aug 29 01:31:37 motioneye meyectl[6950]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 28, in <module>
Aug 29 01:31:37 motioneye meyectl[6950]:     import mediafiles
Aug 29 01:31:37 motioneye meyectl[6950]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 41, in <module>
Aug 29 01:31:37 motioneye meyectl[6950]:     import uploadservices
Aug 29 01:31:37 motioneye meyectl[6950]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 27, in <module>
Aug 29 01:31:37 motioneye meyectl[6950]:     import pycurl
Aug 29 01:31:37 motioneye meyectl[6950]: ImportError: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/
Aug 29 01:31:37 motioneye systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Aug 29 01:31:37 motioneye systemd[1]: motioneye.service: Failed with result 'exit-code'.

```

zagrim, 2019-08-29:
Please try running 
```
pip uninstall pycurl
pip install pycurl
```
to reinstall pycurl which seems to be broken.

el97, 2019-08-29:
@zagrim It seems to work now. Thanks for the help!

closed 2019-08-29 by el97