motioneye-project/motioneye #1435: Picture too bright with high resolution of Pi Camera

cook32, 2019-08-29:
When I am using a resolution upper than 800x600, the picture become really bright and do not return to a normal brightness after seconds.

My config is :

RPI 3
OS Buster
GPU 256
Motioneye 0.40
Motion 4.1.1
Camera Pi V2
V4L2 camera mmal service

Any idea ?

thanks and regards

bfdonny, 2020-03-07:
I'm seeing the same behavior. 

RPI 3 B+
motionEye Version | 0.42
Motion Version | 4.2.2
OS Version | Raspbian 10
SVPRO 1080p Full Hd USB Camera Module (https://www.amazon.com/gp/product/B07C2RL8PB)

I have two of those devices with the same specs (including same camera) and they both experience the same issue. As noted by cook32, resolutions above 800x600 become almost completely washed out. However, I have another RPI 3 B+ running MotionEyeOS with a stock camera module that doesn't experience this issue. Its specs:

RPI 3 B+
motionEye Version | 0.38
Motion Version | 4.1.1
OS Version | motionEyeOS 20180101

tiny0little, 2022-09-08:
same problem

RPi4B
motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Raspbian 11
Pi NOIR camera with fixed color via [vcdbg set awb_mode 0]

problem resolved after motionEye restart

tiny0little, 2022-09-09:
I had the same problem with pi camera. found this problem reported multiple time in different `issues`. as i understand this is outside of motioneye / motion scope to fix it. here's work around for it

I restart motioneye every 20min, restart script considers is motion detection is triggered, and waits

`
$ crontab -l
*/20 * * * * sudo /var/lib/motioneye/restart-motioneye.sh

$ cat /var/lib/motioneye/restart-motioneye.sh
#!/bin/bash

MD_FILE="/var/lib/motioneye/motion-detected"
LR_FILE="/var/lib/motioneye/last_restarted"

meye_restart () {
  echo "restarting ..."
  sudo systemctl restart motioneye.service
  /usr/bin/date "+%Y-%m-%d %H:%M" >> $LR_FILE
  /usr/bin/tail -50 $LR_FILE > $LR_FILE.tmp
  mv $LR_FILE.tmp $LR_FILE
}

if [ ! -f "$MD_FILE" ]
then
  meye_restart
  exit
fi

for i in {1..4}
do
  if test `find "$MD_FILE" -mmin +4`
  then
    meye_restart
    exit
  else
    echo "> try $i"
    echo "motion detection in progress, waiting ..."
    sleep 120
  fi
done

`

this script need to have a line like this
`
touch /var/lib/motioneye/motion-detected
`
or this could be the actual command
[image]

[image]

JamBalaya56562, 2026-11-10:
Closing, as this was with motionEye 0.40-0.42 and Motion 4.1-4.2 on the legacy camera stack, where the camera's exposure did not settle at higher resolutions. Current Raspberry Pi OS runs the camera through libcamera; please open a new issue if pictures are still washed out with a current release.

closed 2026-11-10 by JamBalaya56562