motioneye-project/motioneye #1438: Pictures are doubled (buster)

jartobec, 2019-09-01:
Hello
I have some problems with pictures . (no problem with stretch)
At this times motion works but the picture are doubled , see after (15-50-07**m**.jpeg ?)
[image]
motionEye Version 0.40
Motion Version 4.2.2
OS Version Raspian 10
Raspberry Pi v2 8 MP 1080p Module Caméra
[image]
I.A.W trouble shooting
    • the cable/ribbon/connector ……...-→ **OK**
    • you have loaded the bcm2835-v4l2 kernel module, by adding it to /etc/modules and rebooting afterwards → **not  OK** 
    •  pi@raspberrypi:~ $ sudo apt-get install  bcm2835-v4l2
    • Reading package lists ... Done
       Building the dependency tree
       Reading status information ... Done
       E: Can not find package bcm2835-v4l2
    • In fact folder « modules » does not exist !!
    • you have enabled the camera using raspi-config  → **yes**
    • you have allocated at least 128 MB of RAM to the GPU → **yes**
    • you are using a genuine camera module, not a 
"compatible" one → **yes** (Raspberry Pi v2 8 MP 1080p Module Caméra)
    • the command raspistill …... → **yes if motion is « OFF »**
The special file /dev/video0 should be present when all of the above are met. → **yes**
Make sure you have …. than 0.  → **yes**

motioneye.Log :  `https://pastebin.com/tN5xLRQP`
i have « been warned about   bcm2835_v4l2 » !!
From /var/log « messages » :  `https://pastebin.com/wZbUjfKJ`
Can I have some ideas about  troubleshooting
Thanks you for advance
Kenavo
jartobec

zagrim, 2019-09-02:
The files with names ending with `m` are debug files created be Motion to aid in tuning motion detection parameters. If you don't need them, just turn "Create Debug Media Files" off in the motionEye UI.

jartobec, 2019-09-02:
thanks a lot for your help ;-)
smooth operation
I am stupid .......

closed 2019-09-02 by jartobec