motioneye-project/motioneye #1439: Raspberry Pi zero W fails to start motioneye

Zach0331, 2019-09-02:
Hello,

I have grabbed motioneyeos-raspberrypi-20190427.img from the repo. I flashed it to the SD card using Etcher. I created my wpa_supplicant.conf file and ejected my SD card from my laptop. Then, I inserted it into my pi zero and the camera is installed on it too. I connected a monitor to it and plugged in the power cord. The system finishes everything correctly until it gets to "starting motioneye". That part always fails and reboots the system. I have the camera installed correctly from looking at guides and also comparing to my PI3. I have no clue what else to do. Any help would be greatly appreciated.

silascutler, 2019-09-02:
Hey @Zach0331 

This issue may be better submitted to https://github.com/ccrisan/motioneyeos

JamBalaya56562, 2026-11-08:
Closing, as pointed out above: this was the motionEyeOS image of 20190427, and starting motionEye at boot is part of motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos). Please report it there if it still happens.

closed 2026-11-08 by JamBalaya56562