motioneye-project/motioneye #1440: Add temp overlay or feedback

cpapplefamily, 2019-09-02:
With the Raspberry Pi Zero I observed some hi temperatures in the official pi case. Could  the system collect CPU data like temperature and allow the variable to be in the overlay?

jasaw, 2019-09-02:
This script adds temperature next to camera name in the video.

```
#!/bin/sh

CAM_CONFIG_FILE="/etc/motioneye/Camera-1.conf"
CAM_NAME=''
if [ -f $CAM_CONFIG_FILE ]; then
	CAM_NAME=`grep text_left $CAM_CONFIG_FILE | cut -f2 -d' '`
fi
TEMP=`vcgencmd measure_temp | cut -f2 -d'='`
curl "http://localhost:7999/1/config/set?text_left=$CAM_NAME\%20$TEMP"
```

cpapplefamily, 2019-09-02:
Thanks I have been reading further into the docs and found some mentions on how to do this.  The sample you supplied also hepls.

cpapplefamily, 2019-09-02:
> This script adds temperature next to camera name in the video.
> 
> ```shell
> #!/bin/sh
> 
> CAM_CONFIG_FILE="/etc/motioneye/Camera-1.conf"
> CAM_NAME=''
> if [ -f $CAM_CONFIG_FILE ]; then
> 	CAM_NAME=`grep text_left $CAM_CONFIG_FILE | cut -f2 -d' '`
> fi
> TEMP=`vcgencmd measure_temp | cut -f2 -d'='`
> curl "http://localhost:7999/1/config/set?text_left=$CAM_NAME\%20$TEMP"
> ```

To get me started what Do I do with this? I tried to create the file on the Pi Zero with the camera `/data/etc/monitor_1` but nothing seemed to happen.

jasaw, 2019-09-03:
Are you running motionEyeOS or motionEye + Raspbian ?
If you are running motionEyeOS, you might want to change the `CAM_CONFIG_FILE` to `CAM_CONFIG_FILE="/data/etc/Camera-1.conf"`

You'll need to execute the script to overlay temperature. Something like this.
Run `EDITOR=vi crontab -e`, and add `* * * * * /data/etc/monitor_1` to overlay CPU temperature every minute.

cpapplefamily, 2019-09-03:
@jawsaw thanks for working this out with me. The answer is I'm running both. The pi zero W with the camera is running motioneyeOE but my hub is running Rasbian.

I have been flipping between the two wiki's and attempted to add the file in the appropriate location. At times it seems I can't create it in the pi zero W running motioneyeOS. I read somewhere that the OS versions are write protected to save it from users like me :)

On What hardware do the files need to be added? And vi editor is a new term for me what is that?

jasaw, 2019-09-04:
[vi](https://en.wikipedia.org/wiki/Vi) has been around since 1976. :-)
You can try nano editor, which is more modern.
Check out this guide on how to set up cron. https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support

DaVinci1987, 2020-06-17:
Hi there.
Do this still work with the latest version for pi 4.
Me as a PC type guy like the GUI from MotionEyeOS very much and I like to plot the CPU temp from my pi 4 in the captured video as well. 
I'm not the console typing person and like nice, intuitive GUI's.

Can you tell me, or even show, what and where to change files in order to add a temperatur overlay,

Thank you in advance

Pierreic, 2022-08-25:
Hello !

Thank you Jasaw, It' work perfectly ! :)

But I would like to add colour highlighting (yellow, red or other) and flashing  (see attached).

For example, if CPU is overheating alert and flasher
I tried and i did not succeed...

[image]

Is it possible ? 

Thank you in advance !

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #738, which asks for dynamic values in the text overlay in general; @jasaw's cron script above stays the way to show the temperature until then. Colours and flashing are not possible, as Motion draws the text.

closed 2026-11-13 by JamBalaya56562