motioneye-project/motioneye #1442: [HELP] Turn FNC on/off via terminal/config

dk850, 2019-09-03:
Hi,

I've been searching for hours in the configs and various random ports to see if there is a command/script/config that I can change/run to get the status of the fast network camera and to change it.

What I want to accomplish is to write a function that detects if the Fast Network Camera [FNC] is on, and then it changes it to the opposite (e.g. if its on, turn it off and reboot) but I cant find a single line of code where it turns this FNC on or off. 

The only place I can see it is in the website GUI under "expert" settings. Surely there must be a command I can run to change the state of the FNC, or at least a config file or something that has this option as I imagine its just a simple matter of setting it true or false. I initially thought it would be in the (Device IP):7999/(camera ID)/config/... but obviously that isn't available when you're in FNC mode so I just cant find it at all.

So, can anyone help me find this config option, or a command that turns the Fast Network Camera option ithat is in the GUI on or off but without use of the GUI?

Not sure if this should be help or a feature request or something.

Thanks,
Dean

ccrisan, 2019-09-03:
The following code enables/disables FNC: https://github.com/ccrisan/motioneyeos/blob/dev/board/raspberrypi/motioneye-modules/streameyectl.py#L187

It's more complex than you would expect, because FNC mode is a radically different mode than the regular Motion-based mode. Many things must be reconfigured.

dk850, 2019-09-03:
Thanks for the quick response.

How do you actually switch it using the GUI then? Do you have a seperate control script that you replace the linked file with? 

I imagine it will be fairly easy to change if you have 2 streameyectl.py  One for FNC and one without. Could you point me in the right direction and explain how the GUI button works.

Thanks again

ccrisan, 2019-09-04:
The GUI calls that function highlighted in my previous post.

dk850, 2019-09-04:
So if I just set that to enabled/disabled that will switch the FNC status?

dk850, 2019-09-10:
I've looked using the find command "find / -name *streameyectl.py*" but can't find where about this python script is in the OS (I'm running motioneyeOS on a RPi)

dk850, 2019-09-11:
I've just uncompyled "/usr/lib/python2.7/site-packages/motioneye/streameyectl.pyc" and changed the line of code @ccrisan you mentioned from enabled to disabled and vice versa, then recompiled it every time to see what would happen, and after returning it to its directory and overwriting the file, no FNC has came on after a reboot with either enabled or disabled in use.

JamBalaya56562, 2026-11-08:
Closing, as Fast Network Camera is a motionEyeOS mode, switched by its `streameyectl.py` as ccrisan showed above. motionEyeOS is maintained in its own repository (motioneye-project/motioneyeos), so please ask there.

closed 2026-11-08 by JamBalaya56562