motioneye-project/motioneye #1443: Camera not displaying on MotionEyeOS

SongSpark, 2019-09-04:
Hey all, I set up motioneyeos on a rasberrypi 3b+ and added a NOIR camera to it 
(link to exact camera: https://www.amazon.ca/gp/product/B076KCZRDS/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1)
I've set everything up, the pi is talking to my internet and I can connect to it easily through the IP address, only problem is is that I can't see anything through the camera. 
Screenshot of my view:
![Annotation 2019-09-03 195553]
(https://user-images.githubusercontent.com/54874003/64219982-f270e280-ce84-11e9-9626-2147174d2734.png)
So what am I doing wrong? The physically camera itself has its little red light on, so I'm at least sure I've got it inserted correctly. Suggestions?

el97, 2019-09-04:
Keep in mind that this repository is not for the OS. If you have problems with the OS, ill recommend that you open a issue there. https://github.com/ccrisan/motioneyeos/issues. 
Anyway... Can you test with another camera just to eliminate possible solutions? you can also test the camera with fswebcam: https://www.raspberrypi.org/documentation/usage/webcams/

el97, 2019-09-04:
After some quick scrolling in the issue list I suggest that you look at this issue: #1408

CRImier, 2019-09-26:
Hi! First of all, any chance you're using Google Chrome? It recently updated and broke the technique MotionEye uses for webpage image streaming. You can read #1401 to know more. Tl;dr: you can use another browser that didn't make the same change (i.e. Firefox, Edge, or some Webkit browsers that didn't yet implement this same change), update MotionEye to the latest release, or use the about://flags fix that's described in #1401 .

brettvitaz, 2019-10-08:
FWIW, I had this same issue, and updating to the latest version from the GUI resolved it for me.

JamBalaya56562, 2026-11-07:
Closing as a duplicate of #1401: Chrome 76 broke the way motionEye loaded the live pictures, which b22ef62 fixed in 0.41. As noted above, updating motionEye or using another browser solves it.

closed 2026-11-07 by JamBalaya56562