motioneye-project/motioneye #1444: Motion detected, but no snapshots/videos made anymore

JBS5, 2019-09-05:
Since a few days, no snapshots are made when motion is detected. Motion is detected, as the view turns red. Not using a mask. Also with the mask enabled, no snapshots are made. No movies will be recorded on motion as well.

[image]

No full disk, and the directory is created based on the given name:

[image]

When setting the 'Still Images' settings to 'Interval Snapshots' or 'All frames' instead of motion, the snapshots are made without a problem.

This issues is there for all my cameras

Using motionEye 0.41rc1 and 0.41 in docker on a NUC running Ubuntu Server.
Volumes of the running docker:

[image]

JBS5, 2019-09-06:
Not sure why, but it is working again. Looks like enabling motion detection on another camera forced it to work on the camera with the problem. Any clue why the motion part could crash or not start properly?

JBS5, 2019-12-05:
Installed motionEye on my NUC this way https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu instead of using the docker, and this issue isn't here anymore.

JamBalaya56562, 2026-11-07:
Closing, as the problem went away for you after installing motionEye directly instead of in Docker. The cause stayed unknown; please open a new issue with the motionEye log if it comes back.

closed 2026-11-07 by JamBalaya56562