motioneye-project/motioneye #1445: Images are turning to colorful

Tinke3, 2019-09-05:
Hi, 

I have motioneyeOS dev20190903 (Raspberry PI 4) and two connected Full HD IP cameras (HOSAFE POE IP Camera 1080P). 

After a installation everything is working fine. However, after a couple of minutes all images  (still images, stored videos after a detected motion and streaming) starts to turn colorful. Eventually all pixels are changed, but video is still live picture from camera. If I then make any configuration change in motioneye, the images will return to normal. 

Any ideas?

[image]

[image]

lutin44, 2021-04-11:
Hello same for me, I reduced the resolution of the cameras to 640 * 352, it is a little better but what solution to remain in 1080p and fluid and to have a correct detection

starbasessd, 2021-04-11:
What versions for motion, motioneye, and base os? Using a Pi4?

lutin44, 2021-04-12:
Rapsberry Pi3 B
motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

starbasessd, 2021-04-12:
I would try first going with dev20201026. You also don't mention which camera(s) you are using, and how they are set up, type, URL, etc.

lutin44, 2021-04-12:
I'm using these modele: https://www.amazon.com/Security-Surveillance-Detection-Compatible-Software/dp/B0797KY9JC

connected by rtsp url

af7567, 2021-04-26:
@lutin44 What framerate do you have the camera set to? I have noticed this problem with rtsp cameras which don't support tcp when the framerate is low, but if I put it up to 15+ fps then the problem happens less frequently.

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEyeOS 20190903 and 20200606. Pictures that turn into colourful blocks after a while, and come back when a setting is saved, which restarts Motion, are H.264 frames from the RTSP stream decoded with data lost on the way, usually over UDP. Using TCP for RTSP, a higher frame rate as af7567 found, or the camera's sub stream helps. If it still happens with a current motionEye and Motion, please open a new issue with `motion.log`.

closed 2026-11-08 by JamBalaya56562