motioneye-project/motioneye #1446: ONVIF support

BassT23, 2019-09-07:
### motionEye Version

I am running motionEye version: 0.41

### Board Model

I am using the following board/model: Dell Laptop / Ubuntu Server 18.04

### Camera

I am using the following type of camera: Zeetopin 1080p 
https://www.amazon.de/gp/product/B07N2HQQMS/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

### Network Connection

My motionEyeOS unit is connected to the network via: Cable/Lan

 
I want to add my new ONVIF Camera to motioneye.
4 Cams are online for now (RTSP/Local MMAL/Network IP Cam)
The new one work with TinyCamPro (Android) but cant add to motioneye - nothing will be found :(

Open Ports: 80 and 8000

How could it work?

BassT

noriegafam, 2019-09-18:
If you don't mind, BassT23 how did you get motioneye to install and run in ubuntu server 18.04? I have been following the instructions, and can't get past step 5. Prepare the configuration directory:

$ sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory

Thanks

BassT23, 2019-09-19:
I found the right port via another video software.

this URL work:
rtsp://X.X.X.X:8554/Streaming/Channels/101

closed 2019-09-19 by BassT23