motioneye-project/motioneye #1448: Motioneye continously storering still images

Nostromo-star, 2019-09-08:
Hi all,

I recently set up MotionEye on Debian10 and it is actually running reliable.
Motion triggered events/videos are stored on a NAS drive which is working well.

However, one odd thing is that ME is continously saving still images with 2s intervals to the local
SSD even this function is disabled (OFF) in the related setup.
I need to delete these files manually to avoid filling up the internal SSD although this should not be used at all.

Can anyone give me a hint how to stop this?

many thanks in advance,
Thorsten

Barmatuhin, 2019-09-26:
The same issue here....

Barmatuhin, 2019-09-26:
hm... best than i found to stop filling my disk with jpegs is to set still images on and make them only one for motion....
BUT this is not good fix... 
@ccrisan please look at this issue...

ccrisan, 2019-09-29:
Guys, can you please attach your `camera-x.conf` configuration file so that I can see what's going on? Of course, please remove any sensitive data from it first.

Nostromo-star, 2019-09-29:
Hi ccrisan,

the xxx.conf was an excellent hint from you to look at.

In my case it seems the motion.conf was somehow incorrect. Maybe this was because of my migration when I moved from Raspberry3b+ to an Beelink Z83II.
I used the original Raspberry backup file and this ended up in a double entry in the motion.conf.
[image]

I manually removed the "red" entry at is was exactly the problematic camera channel.
Problem seems to be solved - no further storage of unwanted still images.

Thanks for pointing me to the root cause.

regards,
Thorsten

closed 2019-09-29 by Nostromo-star

Barmatuhin, 2019-09-29:
Here my camera-1.conf (only one camera) and motion.conf (rename them to .txt).
This behavior  starts after i upgrade motioneye from 0.39 to latest 0.41 release.
I found in logs this deprecated parameter #thread and remove it and rename thread-1.conf to this camera-1.conf
Check Mr. Thorsten version of this behavior, but it's not my case... still have a lot of jpegs after motion detected. Forced to use extrascript to remove them before uploading to google drive.

[camera-1.txt](https://github.com/ccrisan/motioneye/files/3667573/camera-1.txt)
[motion.txt](https://github.com/ccrisan/motioneye/files/3667574/motion.txt)

ccrisan, 2019-09-29:
The `thread-*.conf` is indeed a leftover from previous versions. There's a migration script in place but it might have failed in your case during upgrade phase.