motioneye-project/motioneye #1452: Installation on Raspberry Pi 3B+

mitch1138, 2019-09-11:
I have tried to install Motioneye on a fresh install of Raspberry Stretch or Buster on a Raspberry Pi 3 B+ according to https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian but have never been successful - I always end up with a fatal error, most recently summarized here: https://github.com/ccrisan/motioneye/issues/1417#issue-481770471.  Does anyone have a step by step install process that has worked?  Many thanks.

lukdz, 2019-09-16:
Try this method as workaround:
https://github.com/ccrisan/motioneyeos/wiki/Installation
I have successfully installed MotionEye OS on Raspberry Pi 3 B one week ago.

mitch1138, 2019-09-16:
Many thanks.  I also install the MotionEye OS on Raspberry with success.  I will look at the Docker image.  Thanks again.

mitch1138, 2019-09-17:
I am trying the Docker Raspberry pi version but might be having some configuration issues.  After installing Docker and pull the motioneye image (as outlined in the wiki) I started docker using the command below (from the wiki).  It looks to be running without error, but I can't seem to connect using the browser (http://192.168.0.134:8765/).  Did you do something els to get it running on Ubuntu?
Many Thanks, Dave
-------------------------------
docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-amd64

lukdz, 2019-09-18:
I did not change anything from tutorial during install on Ubuntu. 

But I think you should change:
from
> ccrisan/motioneye:master-amd64
to
> ccrisan/motioneye:master-armhf

lukdz, 2019-09-28:
Docker images are static (they can't change by one bit), so they is no reason for they to work once and fail later. 
However you are linking to folders to container, which can be edited and persist data after reboot: 
```
-v /etc/localtime:/etc/localtime:ro
-v /etc/motioneye:/etc/motioneye
-v /var/lib/motioneye:/var/lib/motioneye
```
First on is read only, so it can't be source of the problem. From [Install In Docker](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian):

> Change the two bind paths /etc/motioneye and /var/lib/motioneye according to your needs. The first contains the configuration files for motionEye and the second will be used as file storage for still images and movies. The bound file /etc/localtime is necessary for a proper timezone configuration inside the container using the timezone of the host.

Try:
-stoping container
-deleting content of /etc/motioneye and /var/lib/motioneye
-starting container 
If it works once, and then fails after reboot, try again but reboot without making any changes in motioneye (maybe some config change you make [but don't need] prevents motioneye from rebooting). If it works after "clean" reboot: try making changes once at a time, and rebooting to find what breaks motioneye.

mitch1138, 2019-09-30:
Thank you again for such clear guidance on this issue.  Following your advice, I deleted the contents of /etc/motioneye - after this I was able to reboot multiple times (now up to 3) even after changing the camera configuration.  I will also go back and see if this guidance helps on my issues of getting the standard motioneye installed on the raspberry pi. 

I am closing this and thanks to you successfully.  Thanks again for your unbelievably prompt and accurate guidance.

Dave

closed 2019-09-30 by mitch1138