motioneye-project/motioneye #1454: script for add PTZ?

dimgod59, 2019-09-15:
Hi, 
I would like to install a PTZ system. Can you help me to do the script or the files?

Best.

kleini, 2019-09-17:
What is a PTZ system?

dimgod59, 2019-09-17:
It’s pan tilt device. 2 motors for up/down and left/right plug to GPIO on rpi

nishant2019, 2019-10-12:
Try node red

dimgod59, 2019-10-12:
I don’t know node red :( can you help me ?

nishant2019, 2019-10-12:
> 
> 
> I don’t know node red :( can you help me ?

which is you camera model  ?

dimgod59, 2019-10-12:
It’s a raspberry caméra Night vision (on rpi zéro )

JamBalaya56562, 2026-11-10:
Closing, as motionEye has buttons for this: executable scripts named `/etc/motioneye/up_1`, `down_1`, `left_1` and `right_1` (for camera 1) show arrow buttons on the camera, and each click runs the script, which can drive the motors through the GPIO pins. motionEye does not ship the motor control itself.

closed 2026-11-10 by JamBalaya56562