motioneye-project/motioneye #1455: Bad wiki for Docker ports

lukdz, 2019-09-16:
I came across a problem with [Docker install guide](https://github.com/lukdz/motioneye/wiki/Install-In-Docker) on Ubuntu 18, where Docker would throw error when trying to install motioneye image:

Is:
`-p 8765:8765 8081:8081`
`-p 8765:8765 58081:8081 58082:8082`
Should be:
`-p 8765:8765 -p 8081:8081`
`-p 8765:8765 -p 58081:8081 -p 58082:8082`

Example on [stackoverflow](https://stackoverflow.com/questions/20845056/how-can-i-expose-more-than-1-port-with-docker).

kleini, 2019-09-18:
You edited the page, right? Do we still need this issue then?

lukdz, 2019-09-18:
I edited the page in my [fork](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker), not in [master.ccrisan](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker).

I don't now any way to make pull request for project's wiki on github.

kleini, 2019-09-18:
Sorry, I didn't notice that. Maybe there is a way to create pull requests for the wiki Git. Anyway I added your fixes.

closed 2019-09-18 by kleini