motioneye-project/motioneye #1458: Trying to run as normal user

noriegafam, 2019-09-18:
When issuing command following the instructions:
sudo -u anoriega meyectl startserver -b -c /home/anoriega/motioneye/motioneye.conf

returns:

File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 27, in <module>
    import pycurl
**ImportError: pycurl: libcurl link-time ssl backend (openssl) is different from compile-time ssl backend (none/other)**

remisalmon, 2019-09-18:
This worked for me: https://stackoverflow.com/questions/21096436/ssl-backend-error-when-using-openssl/21099222#21099222

You may need to install `openssl-devel` on Fedora

noriegafam, 2019-09-23:
Only as root was I able to make this work. I found it a few posts bellow the one you linked. Thanks
**su -**
pip uninstall pycurl
export PYCURL_SSL_LIBRARY=openssl
pip install pycurl

closed 2019-09-23 by noriegafam