motioneye-project/motioneye #1459: Are instructions to install in ubuntu correct?

noriegafam, 2019-09-18:
$ pip install motioneye
Successfully installed MarkupSafe-1.1.1 backports-abc-0.5 futures-3.3.0 jinja2-2.10.1 motioneye-0.41 pillow-6.1.0 pycurl-7.43.0.3 singledispatch-3.4.0.3 six-1.12.0 tornado-5.1.1

$ cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory

_When checking there is no '/usr/local/share/motioneye/' foler_

remisalmon, 2019-09-18:
If you installed motioneye in a virtual environment, `motioneye.conf.sample` is located in `/path/to/ENV/motioneye/extra/`

noriegafam, 2019-09-23:
Thank you for responding. I installed ubuntu server on a **non-virtual enviroment** still the same problem. I just can't find where motioneye got installed in ubuntu 18, sorry, this might look like a silly question. This is making me feel like the dumbest Linux user ever. Not being able to find a simple file? I don't understand what is going on! I am the kind of user that searches a lot before posting, however search engines are not helping either

jobenvil, 2019-10-03:
@noriegafam your are not alone:
```
desigual@thor ~ $ sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory
desigual@thor ~ $ sudo find / -name motioneye.conf* -type f
find: ‘/proc/27404’: No such file or directory
desigual@thor ~ $ cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
I realized that I had previously installed motioneye in version 39, but i deleted long time ago the directory. Long history short:

```
sudo pip uninstall motioneye    
sudo pip install motioneye
```

had worked on my side.

noriegafam, 2019-10-03:
Thank you looking into this. Will try it.

GeeAlex, 2019-10-16:
For my 18.04 vm, i found it in .local/share if it helps anyone.

BETLOG, 2020-12-14:
yeah really
sudo -H pip install -U motioneye
seems to solve a bunch of otherwise cryptic pip errors and non-globally located file issues, and service fails

starbasessd, 2020-12-14:
The new instructions are. Many upstream sources are dropping bits and pieces is python 2.7, the latest biggest piece which is still required. Please follow the instructions for Ubuntu 20.04 or Newer for any debian-like Linux including Raspberry Pi

JamBalaya56562, 2026-10-27:
The per-distribution installation pages have been removed from the wiki (December 2025). There is now a single install method for all distributions in the README: https://github.com/motioneye-project/motioneye#installation. `motioneye_init` copies the config sample from the installed package itself, so there is no path to look up anymore. Closing, as the page this refers to no longer exists. If the README instructions do not work for you, please open a new issue.

closed 2026-10-27 by JamBalaya56562