motioneye-project/motioneye #1461: Motion Detection not working with night vision

ubik89, 2019-09-21:
Helo,

I've got a problem.

Everything works fine, just the motion detection not.

If I turn on "Still images" with "All frames" then everything works fine.

When I turn on "Motion Triggered (One Picture)", then nothing seems to be captured.

I tried to shake back and forth through the camera lens, but nothing seems to be captured.

My camera has night vision.

What would be the solution?

ubik89, 2019-09-21:
I'm using it on Raspberry PI Zero W and Raspbian Lite.

gitgoldberg, 2019-09-23:
Your night camera is almost certainly not sensitive to body heat.  You need to illuminate the scene either with a visible-light source or infra-red light before your camera will sense motion.
I've used an infra-red source such as this:
https://www.ebay.co.uk/itm/Infrared-IR-LED-From-Illuminator-Vision-Night-Camera-48-LED-Light-Lamp-850nm-UK/383034548724?epid=756901642&hash=item592ea361f4:g:QdkAAOSwwZtdHWXD
triggered by a PIR attached to an Arduino.  When the PIR detects a moving warm body, the Arduino switches on the light, and the camera then sees this as motion and begins recording.

JamBalaya56562, 2026-11-10:
Closing, as this was answered: Motion detects motion from changes in the picture, so in the dark the scene needs light the camera can see, such as an infrared illuminator, as explained above.

closed 2026-11-10 by JamBalaya56562