motioneye-project/motioneye #1464: rtorrent crashes on starting

alBronko, 2019-09-23:
```
usr1@linux:~$ rtorrent
(10:21:35) Using 'epoll' based polling.
(10:21:35) Using 'epoll' based polling.
(10:21:35) Using 'epoll' based polling.
(10:21:35) Closed XMLRPC log.
Caught Segmentation fault, dumping stack:B] [Port: 51160] [U 0/150] [D 0/150] [
rtorrent(+0x140a9) [0x50d0a9]
linux-gate.so.1(__kernel_sigreturn+0) [0xb7efcd14]
/lib/i386-linux-gnu/libc.so.6(+0x143e95) [0xb7a4ee95]
/usr/local/lib/libxmlrpc_util.so.4(+0x6000) [0xb7cb2000]
Aborted (core dumped)
```

rtorrent 0.9.7-2
libtorrent 0.13.7-1
curl 7.58.0 (i686-pc-linux-gnu) libcurl/7.58.0 OpenSSL/1.1.1c zlib/1.2.11 libidn2/2.2.0 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"

Thanks!

kleini, 2019-09-23:
How is rtorrent related to motionEye?

alBronko, 2019-09-23:
It doesn't. I'm sorry, I have no idea how this was opened under motionEye

closed 2019-09-23 by alBronko